mesh-iit/ergocub-software #193: Launch `yarprobotinterface` manually instead of when spawning the robot in simulation

SimoneMic, 2023-11-20:
Since there are two different versions of the "basic" configuration of the `yarprobotinterface`: namely [ergocub.xml](https://github.com/icub-tech-iit/ergocub-software/blob/master/urdf/ergoCub/conf/ergocub.xml) and [ergocub_ros2.xml](https://github.com/icub-tech-iit/ergocub-software/blob/master/urdf/ergoCub/conf/ergocub_ros2.xml) it would be nice to have the possibility to select which one to launch. Now, the robot interface it's launched when the robot spawns in simulation, via robot [urdf](https://github.com/icub-tech-iit/ergocub-software/blob/001eef320a070e42d956387ef62b18240d688a1c/urdf/ergoCub/robots/ergoCubGazeboV1_minContacts/model.urdf#L3182)

The main issue is that is not possible to launch the ROS2 [remappers](https://github.com/icub-tech-iit/ergocub-software/blob/master/urdf/ergoCub/conf/wrappers/motorControl/all-mc_remapper_ros2.xml)  separately in a different file, at a different time.

For simplicity, I think that removing it from the URDF would be the easiest path. Then the user can launch whichever configuration suits him best via `yarprobotinterface --config <path/to/xml>`

Would this be a feasible option?
I am also open to other solutions, if anything comes to mind.

Thank you

Nicogene, 2023-11-22:
Specifically for running the ROS2 nws a possibility is to run a separate `yarprobotinterface` that, for each part, runs a nwc and a ros2 nws that attach to it. I think that this should be doable for both control boards or sensors/cameras.

This could introduce an overhead in performances (@elandini84 and @randaz81 know it better)

cc @traversaro

randaz81, 2023-11-22:
No, I think this is not a viable solution. The nws should run in the same process space of the device, inside gzserver.

randaz81, 2023-11-22:
The best solution that comes to my mind is to use two different urdfs.
Or you can use a single urdf with a parameter which is replaced, depending on which yarprobotinterface xml is required.

traversaro, 2023-11-22:
Ideally it would be cool to be able to pass some kind of parameter from a Gazebo world, and then launch a different world based on the kind of ergocub you want to run. Would it be a suitable solution for you @SimoneMic ? How are you launching the simulation and/or adding the model to a running simulation?

SimoneMic, 2023-11-22:
Hi @traversaro, I am launching the simulation using this command: `export YARP_CLOCK=/clock && gazebo -s libgazebo_yarp_clock.so -s libgazebo_ros_init.so` (usually I also specify a world file) then I drag and drop the robot from the models menu.

SimoneMic, 2023-11-22:
> 
> Would it be a problem if you inserted the model directly in the world instead of drag and drop manually it?
On my side I don't have any problems

traversaro, 2023-11-22:
Cool, thanks! I try to understand how we can pass parameters to `gazebo_yarp_robotinterface` via the .world file.

traversaro, 2024-01-10:
> Cool, thanks! I try to understand how we can pass parameters to `gazebo_yarp_robotinterface` via the .world file.

Done, a proposal is ready in https://github.com/robotology/gazebo-yarp-plugins/issues/672 .