mesh-iit/ergocub-software #199: Discuss about simplifying collision meshes for the finger links

PasMarra, 2023-12-06:
https://github.com/icub-tech-iit/ergocub-software/assets/143645988/cd19d30b-5f34-4980-926d-d966f296f65c

Hello everyone,
I'm trying to grasp an object with ergoCub in simulation and, as you can see in the above video, the `Real Time Factor` drops to `0.08-0.10 `as soon as the object is grasped.  
Currently, the collision meshes are the same ones used for visual purposes:

https://github.com/icub-tech-iit/ergocub-software/blob/0945da4e405e72b6fa45664a027d0d6fe0fcb008/urdf/ergoCub/robots/ergoCubGazeboV1_1/model.urdf#L675-L678

and this may be the reason of such a high computational burden. 
Would it be possible to discuss on the possibility to replace them with simpler geometrical shapes, e.g. cylinders, or a simplified version with less faces and vertices - that can be obtained, e.g., with Meshlab?

Notice that this substitution was also done on the `iCubGazeboV2_5_visuomanip`

cc: @Nicogene @xEnVrE

traversaro, 2023-12-06:
For reference, we are doing something similar with the feet, see https://github.com/icub-tech-iit/ergocub-software/pull/180 .

traversaro, 2023-12-06:
> Are there instructions on how to run the model generation machinery locally so that we can test that together with @PasMarra after having changed the `yaml` files?

I guess you will need a machine with creo, as now the generation is happening with creo2urdf. @Nicogene probably knows more. However, as a first step you can manually modify the meshes of the link, and you are done with we can look in automatizing the process.

Nicogene, 2023-12-07:
> Thanks for the comment @traversaro , maybe we could try regenerating the model using cylinders with the automated machinery and check if that helps before testing more advanced solutions such as mesh semplification.
> 
> Are there instructions on how to run the model generation machinery locally so that we can test that together with @PasMarra after having changed the `yaml` files?
> 
> Thank you

Yes you need first to run [`creo2urdf`](https://github.com/icub-tech-iit/creo2urdf) first (either from sources or from binaries), and then run it within creo. If you want I can press the button for you for this quick test.

If using cylinders as simplified geometries is not enough @mfussi66 and I could investigate if we can approximate the meshes with convex hulls using Creo APIs or something else, adding an extra option for it

PasMarra, 2023-12-13:
>  If you want I can press the button for you for this quick test.

Hi @Nicogene, unfortunately we don't have creo and we have never used it, so we cannot do the procedure. Could you please test the first option with cilinders for us, so that we can test our pipeline again?

Nicogene, 2023-12-14:
> Hi @Nicogene, unfortunately we don't have creo and we have never used it, so we cannot do the procedure. Could you please test the first option with cilinders for us, so that we can test our pipeline again?

Yes sure, it would be super if you can provide to me the lenght and radius of the cylinder you prefer for the fingers' links, otherwise, I will calculate them measuring on the CAD

PasMarra, 2023-12-14:
> I will calculate them measuring on the CAD

For the time being I don't have a particular preference, so I think that this option is the best..

Nicogene, 2023-12-18:
@PasMarra @xEnVrE I made the changes you requested:

- https://github.com/icub-tech-iit/ergocub-software/pull/205

Could you please see if now when grasping the rtf is ok?

xEnVrE, 2023-12-18:
Yes, @PasMarra is going to test it asap! :)

PasMarra, 2023-12-18:
Hi @Nicogene, I've tested your changes https://github.com/icub-tech-iit/ergocub-software/pull/205#issue-2046733855, but there is something I would like to point out:

- The changes have been applyed only to the left hand, but I need both of them;

[image]

- The cylinders are not aligned with the fingers, so I would ask you to change their pose. By doing so, please be sure that, for each phalanx, the cylinder is tangent to the visual mesh at least on the finger's inner side. 

[image]

PasMarra, 2023-12-18:
@Nicogene

Given that, in the future, the fingertips will play an active role in the grasp, I would also ask you to replace the cylinders with a convex hull as collision mesh. Thank you.

Nicogene, 2023-12-19:
Hi @PasMarra, I realized that I run `creo2urdf` with an unsaved yaml, sorry for the unconvenience.
I committed the right one with the correct urdf, I pushed forced then you have to do pull --rebase.

Right now this is the situation 
[image]

> Given that, in the future, the fingertips will play an active role in the grasp, I would also ask you to replace the cylinders with a convex hull as collision mesh. Thank you.

For this feature, we need to schedule a more consistent activity since this is a feature that is missing in creo2urdf.

Let's see if these simplified collisions allow us to do a decent grasp reducing the rtf, if it doesn't work we will focus on implementing the exportation of the convex hull.

cc @xEnVrE @pattacini

xEnVrE, 2023-12-19:
Thanks for the update @Nicogene. It seems to me that the proximal link of the thumb is still missing a collision volume:

[image]

Is it the case or is it just a visualization glitch? Thank you

Nicogene, 2023-12-19:
> Is it the case or is it just a visualization glitch? Thank you

It was actually an error on the displacement of the `r_thumb_2` collision that for some reason has to have the opposite sign with respect to the others, maybe the link frame is differently oriented?

Anyhow changing the sign of the xyz of the origin fixes the problem.

[image]

xEnVrE, 2023-12-19:
Thanks @Nicogene, I guess @PasMarra will make some tests using the latest URDF and let us know!

Nicogene, 2023-12-19:
Actually, yes there is a problem in the link frame:

### Right thumb

[image]

### Left thumb

[image]

This is not creating problems, just to be aware of it

cc @pattacini @fiorisi @Lawproto

PasMarra, 2023-12-19:
Hi @Nicogene, here is the test.

https://github.com/icub-tech-iit/ergocub-software/assets/143645988/ceac4589-8ef8-4f2d-9398-aad20c1fa6a5

As you can see, the `Real Time Factor` still drops substantially.

I would ask you to adjust the length/pose of the distal phalanx for each finger because they protrude out maybe too much, and to move the proximal phalanx collision mesh towards the hand's inner side since currently it is tangent to the external side.

On my part, I will simplify the bottle's mesh.

Nicogene, 2023-12-19:
> I would ask you to adjust the length/pose of the distal phalanx for each finger because they protrude out maybe too much, and to move the proximal phalanx collision mesh towards the hand's inner side since currently it is tangent to the external side.

Ok for me, it is strange that using cylinders didn't improve the rtf, let's see if with these tweaks it is adjusted a bit

Nicogene, 2023-12-19:
I pushed-forced the reduced collisions (remember to pull --rebase)

[image]

PasMarra, 2023-12-20:
Hi @Nicogene, the `rtf` drops even with the bottle's simplified collision mesh. But I think I've found the reason: the palm. From the previous videos you can see that there is always a direct contact between the object and the palm; so, I've tried a grasp with only fingers, i.e. excluding the palm, and as you can see the problem disappears. For this reason, I think we should find a way to simplify the palm collision mesh.

https://github.com/icub-tech-iit/ergocub-software/assets/143645988/6199ef11-4b38-4e0e-8c97-815ebca79188

Nicogene, 2023-12-20:
Here:

- [`1cc4009` (#205)](https://github.com/icub-tech-iit/ergocub-software/pull/205/commits/1cc40091e1cae7607e8069c54a40b95bbda992de)

I committed the urdf with the palms with simplifyied collisions

[image]

@PasMarra please let me know if now it is fine

PasMarra, 2023-12-20:
Not yet.

[image]

Could you please simplify also these two? At least the left one.

[image]

I think they may be contributing to the problem:

[image]

@Nicogene

PasMarra, 2023-12-20:
As discussed with @Nicogene, we opted for the removal of the two meshes above, since they should not play an important role for the grasp. The problem remains.

[image]

Nicogene, 2023-12-21:
Chatting T2T w/ @PasMarra we concluded that any simplification of the finger/hand mesh did not improve consistently the rtf, we noticed that also in the `iCub` model there is the same rtf drop when starting grasping, with the difference that when it used the `iCub` model we start from an higher rtf.

Then it makes more sense to work on increasing the rtf when using `ergoCub` model overall, I will open a dedicated issue for that.
For all those reasons I will not merge:
- #205

commit 4a3ceb91 (2023-12-19): ergoCubGazeboV1_1: simplify collision geometry of the fingers

For augmenting the real time factor we used to simplify the collision geometry of the fingers with cylinders. The measures have been taken roughtly from the CAD. It fixes #199
commit 89e3cacc (2023-12-19): ergoCubGazeboV1_1: simplify collision geometry of the fingers

For augmenting the real time factor we used to simplify the collision geometry of the fingers with cylinders. The measures have been taken roughtly from the CAD. It fixes #199
commit b6058328 (2023-12-19): ergoCubGazeboV1_1: simplify collision geometry of the fingers

For augmenting the real time factor we used to simplify the collision geometry of the fingers with cylinders. The measures have been taken roughtly from the CAD. It fixes #199
commit e61dd1de (2023-12-19): ergoCubGazeboV1_1: simplify collision geometry of the fingers

For augmenting the real time factor we used to simplify the collision geometry of the fingers with cylinders. The measures have been taken roughtly from the CAD. It fixes #199