mesh-iit/ergocub-software #200: [ergoCubSN001] Reenable FT sensors in URDF

pattacini, 2023-12-06:
## Task description
Following up on https://github.com/robotology/robots-configuration/pull/598#issuecomment-1841606958, we're requested to update the URDF of ergoCubSN001 and reinstate the FT sensors.

## Definition of Done
- URDF modified accordingly
- URDF tested in simulation

cc @Nicogene @GiulioRomualdi

Nicogene, 2023-12-07:
The arms FTs seems already there:

https://github.com/icub-tech-iit/ergocub-software/blob/54e090910392177c176c1c1748bb02745fc1191b/urdf/ergoCub/robots/ergoCubSN001/model.urdf#L3262-L3275

AntonioConsilvio, 2023-12-07:
Hi @Nicogene @pattacini, I think there was a little misunderstanding. The FTs on the shoulders have been disabled from configuration files due to malfunctioning, so the sensors should be removed from the URDF to be compatible with the configuration files.

pattacini, 2023-12-07:
Ok thanks @AntonioConsilvio for the heads-up 👍🏻 
I've update the OP and the title accordingly.

Nicogene, 2023-12-07:
@GiulioRomualdi do you need this change in `ergoCubSN001` or/and `ergoCubGazeboV1` urdf?

Nicogene, 2023-12-20:
PR:

- #206

GiulioRomualdi, 2023-12-20:
> @GiulioRomualdi do you need this change in `ergoCubSN001` or/and `ergoCubGazeboV1_1` urdf?

Sorry, I completelly missed the message :( 
We need to remove them from `ergoCubSN001` no need from `ergoCubGazeboV1_1`

Nicogene, 2023-12-20:
> > @GiulioRomualdi do you need this change in `ergoCubSN001` or/and `ergoCubGazeboV1_1` urdf?
> 
> Sorry, I completely missed the message :( We need to remove them from `ergoCubSN001` no need from `ergoCubGazeboV1_1`

I am afraid that I have to remove it also from `ergoCubGazeboV1_1` since I have to remove the nws for those ft from the xml that is common between all the models.

But if the model `ergoCubSN001` is used without launghing the yarprobotinterface plugin I can remove them only from that urdf an keep the ergocub.xml unchanged.

cc @traversaro

GiulioRomualdi, 2023-12-20:
In this case i think we should update wbd  xml file. Are there 2 separate files for gazebov1 and gazebov1_1?

traversaro, 2023-12-21:
Related to https://github.com/robotology/whole-body-estimators/issues/32, that should simplify doing this.

Nicogene, 2023-12-21:
> In this case i think we should update wbd xml file. Are there 2 separate files for gazebov1 and gazebov1_1?

Unfortunately not, it was not needed until now, but I would avoid adding an xml file for this temporary situation

commit 995932fe (2023-12-20): ergoCubV1_1: remove ft sensors of the arms

It fixes #200
commit 6be58932 (2023-12-28): ergoCubV1_1: remove ft sensors of the arms

It fixes #200
PR #206 (2023-12-28): ergoCubV1_1: remove ft sensors of the arms
It fixes #200

As side effect this PR removes also the nws for ergocub 1.0, the optimal solution would be to have two different xml files (e.g. `ergocub1_0.xml`. `ergocub1_1.xml`.

But since this a temporary workaround that will be hopefully soon reverted we could avoid to add yet another xml file to be maintained.

What do you think about @traversaro @GiulioRomualdi @pattacini ?
closed 2023-12-28 by Nicogene