mesh-iit/ergocub-software #201: [ergoCubSN001] no time synchronization between robot and laptop

SimoneMic, 2023-12-07:
## Task description
As emerged in these issues: #142 and https://github.com/robotology/icub-tech-support/issues/1611
The `ergocub-torso` is lagging behind of around 77 seconds.
I have to manually sync using `ntpdate`, however that's not present on the head.

It would be nice to have an automatic sync like on https://github.com/robotology/icub-tech-support/issues/1611 and to port it to all the future robots.

## Definition of Done
Sync the three PCs with the server `10.0.2.1` at startup/boot

cc @davidelasagna

SimoneMic, 2024-01-22:
Is there any possibility to implement this on the robot as it was done on `ergoCubSN00` ?
Or do I have to open a ticket in another place?

Thanks

Nicogene, 2024-01-23:
I opened an internal ticket, we will address it there. Closing

closed 2024-01-23 by Nicogene