mesh-iit/ergocub-software #202: make install do not install the modified models if cmake is note called in advance

GiulioRomualdi, 2023-12-11:
I noticed that if I don't call `cmake .` `make install` does not install the models

cc @traversaro

Nicogene, 2023-12-12:
> I noticed that if I don't call `cmake .` `make install` does not install the models
> 
> cc @traversaro

If I am not mistaken it is not possible to do make without configuring it afterwards. or at least you can but it will do nothing

traversaro, 2023-12-12:
The problem is that all the copy operation from the source to the build directory are done at CMake configure time (i.e. when `cmake` is called, see https://github.com/icub-tech-iit/ergocub-software/blob/91efee5df4ba9de343adce96607bcf1f7c303e5f/urdf/CMakeLists.txt#L30). To copy files at build time (i.e. when `make` is called) one should define custom target to do the copy (see for example https://discourse.cmake.org/t/copying-config-specific-paths-at-build-time/4776/4), ideally defining appropriately the file on which the copy depends. I am not sure if the additional complexity at the CMake code level is worth the gain in usability.