mesh-iit/ergocub-software #203: Discuss the possibility of integrating whole-body dynamics into the robot interface of the simulation

GiulioRomualdi, 2023-12-18:
I  am opening this issue to discuss the possibility of enabling the automatic execution of whole-body dynamics when the robot is dropped into Gazebo.

In our research at AMI, we utilize whole-body dynamics to calculate the estimated contact force. Currently, many team members manually run whole-body dynamics after placing the robot in Gazebo. Automating this process would simplify the simulation.

Furthermore, WBD running within the same YARP robot interface as the simulation would eliminate the need for a network wrapper to retrieve data from the simulated robot.

cc @traversaro @Nicogene

traversaro, 2023-12-18:
Related issue: https://github.com/icub-tech-iit/ergocub-software/issues/35 .

Nicogene, 2023-12-18:
Hi @GiulioRomualdi, do you mean making run wbd via the yarprobotinterface gazebo plugin?
This may be a problem for whom wants to use ergoCub model without having installed the `whole-body-estimators` repo, this is a similar discussion we had about putting the ros2 nws spawned by default

GiulioRomualdi, 2023-12-18:
Yes that's make sense you're right

GiulioRomualdi, 2023-12-20:
"One possibility is to verify whether the 'whole-body-estimator' is installed in the system. If it is, the XML file can then include the device. Otherwise, this functionality can be enabled through CMake."

traversaro, 2023-12-20:
> One possibility is to verify whether the `whole-body-estimator` is installed in the system. If it is, the XML file can then include the device. Otherwise, this functionality can be enabled through CMake.

To be honest, I would avoid that, as then downstream you need to handle both cases, and it may be quite easy to get the wrong file. As most users of ergocub are inside IIT, I would just understand if that is actually a problem for someone, and if not we just enable whole-body-estimator, what do you think?

Nicogene, 2023-12-20:
> To be honest, I would avoid that, as then downstream you need to handle both cases, and it may be quite easy to get the wrong file. As most users of ergocub are inside IIT, I would just understand [...]

I think that this is something that can be asked in the internal channel of ergoCub, if everyone agrees we can add it

traversaro, 2024-01-10:
We could use https://github.com/robotology/gazebo-yarp-plugins/issues/672 to either permit to disable or to enable (depending on what we want the default to be) wbd. If the default on the robot is to have wbd on, I would keep it by default on also on the simulated robot.

GiulioRomualdi, 2025-02-18:
It would be amazing!

Nicogene, 2025-02-20:
> I think we can go back to this once [robotology/gz-sim-yarp-plugins#238](https://github.com/robotology/gz-sim-yarp-plugins/pull/238) is merged. In particular, I would do the following:
> 
>     * Enable WBD by default, with a disabled_by tag that permits users to disable it
> 
>     * Add ROS2 wrappers, but with a enabled_by tag that permits users to enabled them, but by default they will be disabled
> 
> 
> In this way we would align the real robot and the simulated robot, what do you think?

I opened an issue for using that feature!
- #288 

My dream is to share the XML files w/ the ones of the real robot, probably is still to soon