mesh-iit/ergocub-software #213: ergoCubEmotions uses more than 100% of the cpu on ergocub-head

GiulioRomualdi, 2024-01-18:
I noticed that `ergoCubEmotions` uses a really high amount of CPU on the ergocub-head.

In the following screenshot you can check the amount of used CPU (181%) only ergoCubEmotions was running on the ergocub-head
[image]

I was wondering if it is normal

cc @martinaxgloria

traversaro, 2024-01-18:
If I remember correctly once we checked that, one problem may be that at every module update, the file are re-read (see https://github.com/icub-tech-iit/ergocub-software/blob/eaacbf120b355edd931d2ae272e568b507740d2e/src/modules/ergoCubEmotions/ergoCubEmotions.cpp#L119). Perhaps we could read the images only once in the configure, and save them as cv::Mat attributes that can be used in the updateModule?

martinaxgloria, 2024-01-22:
Hi @traversaro, 

thank you for your suggestion. Actually, we are reading videos and not images (the _static_ part is a 2 sec looped video and the transitions from one expression to another too). If I remember well, opencv shows videos one frame at the time after they are opened. I don't know if it's possible to save cv::VideoCapture elements as per cv::Mat ones. 

Today I tried to use the configuration `image -> transition -> video` and I saw from jtop that the usage is reduced to ~30% but there's a spike when the transition is called. 

Moreover, the % showed by jtop should correspond to the sum of the cpu usage for each core: in fact running `top` with IRIX mode set to OFF, this is the cpu consumption divided by the number of cores:

[image]

This means that the cores should not be saturated, I don't know if you see some delays.
I'm going to understand if it's possible to prevent the videos from being reopened at each update 

cc @Nicogene @GiulioRomualdi

martinaxgloria, 2024-01-22:
I opened a draft PR with the suggestions of @traversaro.

- https://github.com/icub-tech-iit/ergocub-software/pull/214

It's in draft since I have to test it on ergocub-head

cc @Nicogene @GiulioRomualdi

Nicogene, 2024-01-23:
The @martinaxgloria fix improve significantly the performance, @GiulioRomualdi in case not sufficient feel free to reopen it.

closed 2024-01-23 by Nicogene