mesh-iit/ergocub-software #216: Add FT IMU in the ergoCubGazeboV1* models

GiulioRomualdi, 2024-02-12:
## Description of the Task
The robot's Force/Torque (FT) sensor is equipped with an Inertial Measurement Unit (IMU) capable of streaming orientation data. Integrating this feature into simulation would be beneficial.

Previously, @G-Cervettini initiated a Pull Request (PR) to integrate this functionality, https://github.com/icub-tech-iit/ergocub-software/pull/145. However, it involved manual modifications to the urdf file.

## Definition of Completion
Successfully implement the IMU streaming within the Gazebo model using multipleanalogsensorserver.

Nicogene, 2024-02-12:
@martinaxgloria also will benefit of this change

GiulioRomualdi, 2024-02-12:
I'm going to open a PR for the feet imu (added with xml blob)

Nicogene, 2024-02-12:
> I'm going to open a PR for the feet imu (added with xml blob)

Why not as `sensors` in the yaml? Using the same CSYS of the ft gives problems?

GiulioRomualdi, 2024-02-12:
I've never did it. Can you point me some line of the code I can use as inspiration?

GiulioRomualdi, 2024-02-12:
Probably something similar to

https://github.com/icub-tech-iit/ergocub-software/blob/master/urdf%2Fcreo2urdf%2Fdata%2Fergocub1_0%2FERGOCUB_all_options.yaml#L808

Nicogene, 2024-02-12:
At the end in :

- https://github.com/icub-tech-iit/ergocub-software/pull/217

We went for adding as xml blobs because we spotted a bug in creo2urdf:

- https://github.com/icub-tech-iit/creo2urdf/issues/77

The generated imu sensors had wrong poses, in `simmecanics_to_urdf` seems not to be a problem exporting an ft and imu that refers to the same csys:

- https://github.com/robotology/icub-models/commit/e9d33b61f36955a8da238c2c06389816c3c1ec85

closed 2024-02-13 by Nicogene