mesh-iit/ergocub-software #221: ergoCubSN001 gravity compensation / compliant mode does not work as intended

traversaro, 2024-02-13:
## Task description

While debugging torque control with @LoreMoretti and @danielePucci, we noticed that switching on compliant mode of the upper body of the ergoCubSN000 the gravity compensation was not working as intended, i.e. by decreasing the stiffness the arm or the torso the joint was not remaining still.  Interestingly, running a gravity compensation "externally" (for example via https://github.com/vvv-school/tutorial_impedance-control/) seems to work fine, indicating that somehow the gravity compensation computed by wholebodynamics device was not working correctly. As both methods used the same model, perhaps the difference may be in the source of kinematic information used by WBD? If anyone is interested in investigating this, a first test could be to change this. From the configuration file in https://github.com/icub-tech-iit/ergocub-software/blob/10670acf8120bdaad19c6cb01388eff555f4f356/urdf/ergoCub/conf/estimators/wbd_ecub_sim.xml#L10C22-L10C34 it seems that wbd is using the head IMU, but according to @GiulioRomualdi this is not the case anymore, so the first thing to do would be to double check this.

Unfortunately I do not have detailed software version information of the software used in the tests, perhaps @LoreMoretti has more info.

## Definition of Done

Commit in master branch a wbd configuration file for which compliant mode works "fine", i.e. by decreasing the stiffness the robot keep its position.

GiulioRomualdi, 2024-02-13:
Here is the bug https://github.com/robotology/robots-configuration/blob/devel/ergoCubSN001%2Festimators%2Fwholebodydynamics.xml#L10

The imu used is the root link imu but the link is the head one

traversaro, 2024-02-13:
> Here is the bug https://github.com/robotology/robots-configuration/blob/devel/ergoCubSN001%2Festimators%2Fwholebodydynamics.xml#L10
> 
> The imu used is the root link imu but the link is the head one

Great, thanks!

traversaro, 2024-02-13:
Fix proposed in https://github.com/robotology/robots-configuration/pull/614 . Note that probably this did not only affected the gravity compensation torques, but also the torque estimation.

isorrentino, 2024-02-14:
@traversaro I checked the configuration I was using for the paper to be sure that the results were correct and it was correct (luckily). Indeed I set the parameter assume_fixed and the frame was root_link.

See:
https://github.com/ami-iit/robots-configuration/blob/4edd9f0993c9e1bf31e3652d67fc7b5d6c38b4ac/ergoCubSN000/estimators/wholebodydynamics.xml#L10

traversaro, 2024-02-14:
> @traversaro I checked the configuration I was using for the paper to be sure that the results were correct and it is correct (luckily). Indeed I set the parameter assume_fixed and the frame was root_link.
> 
> See: https://github.com/ami-iit/robots-configuration/blob/4edd9f0993c9e1bf31e3652d67fc7b5d6c38b4ac/ergoCubSN000/estimators/wholebodydynamics.xml#L10

Great!

Nicogene, 2024-02-16:
@traversaro is this issue fixed by https://github.com/robotology/robots-configuration/pull/614 or we need to change also the urdf?

traversaro, 2024-02-16:
In theory no change should be done in the URDF, in practice until this is tested is difficult to be 100% sure.

LoreMoretti, 2024-03-01:
> This test has been done? Can we close it?

We tested today as per [this comment](https://github.com/ami-iit/element_qp-reactive-control/issues/72#issuecomment-1973555967).

I think we can close this issue.

traversaro, 2024-03-01:
Actually this issue was to track the switch to compliant mode via yarpmotorgui, I am not sure this was tested today. I expect it to work, but I think we can quickly check this on Monday.

Nicogene, 2024-03-18:
> Actually this issue was to track the switch to compliant mode via yarpmotorgui, I am not sure this was tested today. I expect it to work, but I think we can quickly check this on Monday.

@traversaro @LoreMoretti did you checked it?

traversaro, 2024-03-18:
I did not, today there are some demos, we can check after the demos.