mesh-iit/ergocub-software #223: couplingXCubHandMk5 – Refactor the param parsing for using yarpDeviceParamParserGenerator

Nicogene, 2024-02-23:
## Task description

`yarpDeviceParamParserGenerator` is a new YARP  functionality that allows to automatically generate the parsing of the paramenter and documentation for device drivers:

- https://github.com/robotology/yarp/discussions/3081

## Definition of Done

The parse of the parameter has been ported

Nicogene, 2024-03-26:
I did the required changes in this PR:
- https://github.com/icub-tech-iit/ergocub-software/pull/231

Unfortunately, the merging is blocked because we need to contextually update of the controlboard gyp for parsing the lists using `find` instead of `findGroup`

cc @randaz81 cc @pattacini

Nicogene, 2024-04-04:
> > Unfortunately, the merging is blocked because we need to contextually update of the controlboard gyp for parsing the lists using find instead of findGroup
> 
> I haven't understood if I should do something in this respect?

@traversaro made already the changes in this PR:

- https://github.com/robotology/gazebo-yarp-plugins/pull/677

But probably we need to change all the ini of the controlboard for adding the parenthesis?

traversaro, 2024-04-04:
> But probably we need to change all the ini of the controlboard for adding the parenthesis?

Yes, it would be ideal if we could wait for https://github.com/robotology/gz-sim-yarp-plugins/issues/138, I think we can do it in a few days and do a release of gz-sim-yarp-plugins after that.

closed 2024-12-16 by Nicogene