mesh-iit/ergocub-software #228: Add ergoCub SN002 + ergoCubGazeboV1_2 (?) urdf

Nicogene, 2024-03-19:
## Task description
Since `ergoCub` SN002 is close to being delivered we need the urdf of the new robot. Respect SN001 from the urdf side it should be identical, the only difference is in the head's weight.
Probably the version will be increased to V1_2, we need to update also the gazebo models.

## Definition of Done
The new urdfs have been added.

pattacini, 2024-03-19:
cc @Fabrizio69

Nicogene, 2024-04-17:
Adding it should be easier, see:

- https://github.com/icub-tech-iit/ergocub-software/issues/235#issuecomment-2061350998

Nicogene, 2024-04-18:
First step:
- https://github.com/icub-tech-iit/ergocub-software/pull/241

Nicogene, 2024-04-18:
See:
- https://github.com/icub-tech-iit/ergocub-software/pull/242

PR #243 (2024-04-18): Add SN002/V1_2 to unit tests
As per title.

It fixes #228
closed 2024-04-18 by Nicogene