mesh-iit/ergocub-software #239: ergoCubSN002 missing

GiulioRomualdi, 2024-04-17:
## Task description
Today ergoCubSN002 was released however the model is not available. I think it is similar to ergoCUbSN001

## Definition of Done
Having the model in the repo cc @Nicogene and @traversaro

Nicogene, 2024-04-17:
> Today ergoCubSN002 was released however the model is not available. I think it is similar to ergoCUbSN001

Yes probably only the assignedMass and `assignedInertia` has to be changed for `head` link.

Isn't it this issue a duplication of #228 ?

GiulioRomualdi, 2024-04-17:
> Isn't it this issue a duplication of https://github.com/icub-tech-iit/ergocub-software/issues/228 ?

Ops my fault! I'm sorry for that :(

closed 2024-04-17 by GiulioRomualdi

Nicogene, 2024-04-17:
> > Isn't it this issue a duplication of #228 ?
> 
> Ops my fault! I'm sorry for that :(

No problem!