mesh-iit/ergocub-software #244: Align the xml files in order to use enabled_by and disabled_by

Nicogene, 2024-04-22:
## Task description
We need to align the xml files of the simulated robot to the real ones, now the real robot uses the `enabled_by` and `disabled_by` clauses, see:
- Cleanup yarprobotinterface conf files for ergoCub

## Definition of Done

The xml are aligned

cc @traversaro @pattacini

traversaro, 2024-04-22:
The problem is that for simulated robot at the moment you have no way to pass the `enable_tags` and `disable_tags` values in `gazebo-yarp-controlboard` plugin, so before doing this we need to appropriately modify the plugins, see:
* https://github.com/robotology/gazebo-yarp-plugins/issues/672
* https://github.com/robotology/gz-sim-yarp-plugins/issues/163

Nicogene, 2024-04-22:
> The problem is that for simulated robot at the moment you have no way to pass the `enable_tags` and `disable_tags` values in `gazebo-yarp-controlboard` plugin, so before doing this we need to appropriately modify the plugins, see:
> 
>     * [Ensure that enable_tags and disable_tags options can be passed in a world SDF file to the gazebo_yarp_robotinterface of an included model robotology/gazebo-yarp-plugins#672](https://github.com/robotology/gazebo-yarp-plugins/issues/672)
> 
>     * [Ensure that enable_tags and disable_tags options can be passed in a world SDF file to the gazebo_yarp_robotinterface of an included model robotology/gz-sim-yarp-plugins#163](https://github.com/robotology/gz-sim-yarp-plugins/issues/163)

Yes thanks @traversaro I forgot about it, I put it back in the backlog awaiting that the feature in gyp will be available 👍🏻

traversaro, 2024-04-22:
> Yes thanks @traversaro I forgot about it, I put it back in the backlog awaiting that the feature in gyp will be available 👍🏻

Just to be aligned, at the moment no one is planning/scheduling to work on that issue.