mesh-iit/ergocub-software #245: Inconsistent frames in the left/right thumb of ergoCubSN001/2 urdf

giotherobot, 2024-04-29:
Hello, I have found some small inconsistencies in the frame definition of the thumbs in the urdf.

# Right hand
[image]

# Left hand
[image]

The inconsistencies are:
1) Wrt all the other fingers the thumb is the only one where the x axis is the one in the direction of the finger, instead all other fingers have the z axis in the direction of the finger;
2) Between the two hands, if the X axis is in the direction of the thumb, the right hand has the three frames oriented forward, backward, forward and the left hand backward, forward, forward;
3) There is a 90deg rotation of the first frame of the thumb wrt the other frames, around the X axis, that is not present in all other fingers.

cc @GiulioRomualdi

mebbaid, 2024-04-29:
cc @ehsanRanjbari

Nicogene, 2024-05-02:
Hi @giotherobot thanks for reporting the issue.

We will tackle it as soon as the new Program Increment starts (~ mid-May, cc @pattacini).

I can confirm that the urdf correctly represents how the frames are defined in the mech assembly

## Left hand
[image]

## Right hand

[image]

> There is a 90deg rotation of the first frame of the thumb wrt the other frames, around the X axis, that is not present in all other fingers.

This is for sure a bug in the assembly.

>Wrt all the other fingers the thumb is the only one where the x axis is the one in the direction of the finger, instead all other fingers have the z axis in the direction of the finger;

>Between the two hands, if the X axis is in the direction of the thumb, the right hand has the three frames oriented forward, backward, forward and the left hand backward, forward, forward;

For these two latter points, I would cc @Lawproto that worked on this assembly

Lawproto, 2025-07-16:
@MeliTaliani as per our F2F chat, proceed as follows:
- For the two phalanges of the left hand, rotate the frame along the green y-axis by 90° so that the blue z-axis points towards the center of the palm.
  - `L_HAND_THUMB_1`
  - `L_HAND_THUMB_2`
- Rinse and repeat for the right hand (blue towards the center of the palm for reference).
  - `R_HAND_THUMB_1`
  - `R_HAND_THUMB_2`
- Have fun.

Feel free to contact me if you need help.

[image]

Nicogene, 2025-07-17:
I tried to generate the urdf, the urdf of SN001 it works fine, the SN002 instead fail, due to probably this problem:

[image]

Lawproto, 2025-07-17:
> I tried to generate the urdf, the urdf of SN001 it works fine, the SN002 instead fail, due to probably this problem:

At this point, I would suggest to repair all the assemblies that use the parts that have been recreated. At the moment, I see these others:
- `sim_ergocub_1-3.asm.1`
- `sim_cer_emorph.asm.1`

MeliTaliani, 2025-07-17:
Yes, I checked and repaired` sim_ergocub_1-3.asm.1`.` sim_cer_emorph.asm.1` it was already correct.

Lawproto, 2025-07-17:
> Yes, I checked and repaired` sim_ergocub_1-3.asm.1`.` sim_cer_emorph.asm.1` it was already correct.

@MeliTaliani you are right, because unfortunately these pieces has been given new names for R1 even though they are identical to ergoCub's. Alas, this means that they suffer from the same inconsistencies you just repaired for their siblings.

[image]

I propose this quick fix:
- Create copies of the four phalanges you adjusted.
- Rename them with the `SIM_EMORPH` prefix so as to overwrite the existing CER ones.
- Repair ` sim_cer_emorph.asm.1`.