mesh-iit/ergocub-software #246: Unable to launch `launch_wholebodydynamics_ecub.xml` in simulation

SimoneMic, 2024-05-02:
Hello,

I am unable to launch the wholebodydynamics in gazebo, with the current `robotology-superbuild` and `ergocub-software` master branches.

While launching the YRI, I am getting this warning:
``` [DEBUG] wholeBodyDynamics Statistics: Opening contact frames.
[WARNING] subModel no  0  has the maximum number of variables (6). The frame  r_hand_palm  will not be added to it.
[WARNING] subModel no  0  has the maximum number of variables (6). The frame  root_link  will not be added to it.
[WARNING] subModel no  0  has the maximum number of variables (6). The frame  l_foot_front  will not be added to it.
[WARNING] subModel no  0  has the maximum number of variables (6). The frame  l_foot_rear  will not be added to it.
[WARNING] subModel no  0  has the maximum number of variables (6). The frame  r_foot_front  will not be added to it.
[WARNING] subModel no  0  has the maximum number of variables (6). The frame  r_foot_rear  will not be added to it.
[WARNING] subModel no  0  has the maximum number of variables (6). The frame  l_upper_leg  will not be added to it.
[WARNING] subModel no  0  has the maximum number of variables (6). The frame  r_upper_leg  will not be added to it.
```
And I think that it leads to this following error:
```
[ERROR] Model :: getLinkIndex : Impossible to find link l_foot_front in the Model
[ERROR] wholeBodyDynamics : Link  l_foot_front  not found in the model.
[ERROR] wholeBodyDynamics: Problem in opening external wrenches port.
[ERROR] |yarp.dev.PolyDriver|wholebodydynamics| Driver <wholebodydynamics> was found but could not open
```

Hover, the frames are correctly present in the urdf, and the `YARP_ROBOT_NAME` variable is properly set.

I have exactly no clue on what could be the solution here.

Here's the full log for completion:
```
ecub_docker@simone-Alienware-m16-R1:~$ launch_wbd_interface 
[ERROR] |yarp.os.Property| cannot read from yarprobotinterface.ini
[DEBUG] Reading file /home/ecub_docker/robotology-superbuild/src/ergocub-software/urdf/ergoCub/conf/launch_wholebodydynamics_ecub.xml
[WARNING] Invalid syntax while loading /home/ecub_docker/robotology-superbuild/src/ergocub-software/urdf/ergoCub/conf/launch_wholebodydynamics_ecub.xml at line 1 . Expected document of type robot . Found devices
[DEBUG] yarprobotinterface: using xml parser for DTD v3.x
[DEBUG] Reading file /home/ecub_docker/robotology-superbuild/src/ergocub-software/urdf/ergoCub/conf/launch_wholebodydynamics_ecub.xml
[INFO] Yarprobotinterface was started using the following enable_tags: 
[INFO] Yarprobotinterface was started using the following disable_tags: 
[DEBUG] List of all enable attributes found in the include tags: 
[DEBUG] List of all disable attributes found in the include tags: 
[DEBUG] Preprocessor complete in:  8.91685e-05 s
[WARNING] Invalid syntax while loading  at line 4 . "robot" element should contain the "portprefix" attribute. Using "name" attribute
[WARNING] *************************************************************************************
* yarprobotinterface 'portprefix' parameter does not follow convention,  *
* it MUST start with a leading '/' since it is used as the full prefix port name    *
*     name:    full port prefix name with leading '/',  e.g.  /robotName      *
* A temporary automatic fix will be done for you, but please fix your config file   *
*************************************************************************************
[INFO] |yarp.os.Port|/ergoCubGazeboV1/yarprobotinterface| Port /ergoCubGazeboV1/yarprobotinterface active at tcp://10.0.3.1:10053/
[INFO] startup phase starting...
[INFO] Opening device torso_mc with parameters [("robotName" = "ergoCubGazeboV1"), ("remote" = "/ergocubSim/torso"), ("local" = "/wholeBodyDynamics/torso")]
[DEBUG] |yarp.dev.PolyDriver|torso_mc| Parameters are (device remote_controlboard) (id torso_mc) (local "/wholeBodyDynamics/torso") (remote "/ergocubSim/torso") (robotName ergoCubGazeboV1)
[INFO] |yarp.os.Port|/wholeBodyDynamics/torso/rpc:o| Port /wholeBodyDynamics/torso/rpc:o active at tcp://10.0.3.1:10054/
[INFO] |yarp.os.Port|/wholeBodyDynamics/torso/command:o| Port /wholeBodyDynamics/torso/command:o active at tcp://10.0.3.1:10055/
[INFO] |yarp.os.Port|/wholeBodyDynamics/torso/stateExt:i| Port /wholeBodyDynamics/torso/stateExt:i active at tcp://10.0.3.1:10056/
[INFO] |yarp.os.impl.PortCoreOutputUnit|/wholeBodyDynamics/torso/rpc:o| Sending output from /wholeBodyDynamics/torso/rpc:o to /ergocubSim/torso/rpc:i using tcp
[INFO] |yarp.os.impl.PortCoreOutputUnit|/wholeBodyDynamics/torso/command:o| Sending output from /wholeBodyDynamics/torso/command:o to /ergocubSim/torso/command:i using udp
[INFO] |yarp.os.impl.PortCoreInputUnit|/wholeBodyDynamics/torso/stateExt:i| Receiving input from /ergocubSim/torso/stateExt:o to /wholeBodyDynamics/torso/stateExt:i using udp
[INFO] |yarp.dev.PolyDriver|torso_mc| Created device <remote_controlboard>. See C++ class RemoteControlBoard for documentation.
[INFO] Opening device left_arm_mc with parameters [("robotName" = "ergoCubGazeboV1"), ("remote" = "/ergocubSim/left_arm"), ("local" = "/wholeBodyDynamics/left_arm")]
[DEBUG] |yarp.dev.PolyDriver|left_arm_mc| Parameters are (device remote_controlboard) (id left_arm_mc) (local "/wholeBodyDynamics/left_arm") (remote "/ergocubSim/left_arm") (robotName ergoCubGazeboV1)
[INFO] |yarp.os.Port|/wholeBodyDynamics/left_arm/rpc:o| Port /wholeBodyDynamics/left_arm/rpc:o active at tcp://10.0.3.1:10057/
[INFO] |yarp.os.Port|/wholeBodyDynamics/left_arm/command:o| Port /wholeBodyDynamics/left_arm/command:o active at tcp://10.0.3.1:10058/
[INFO] |yarp.os.Port|/wholeBodyDynamics/left_arm/stateExt:i| Port /wholeBodyDynamics/left_arm/stateExt:i active at tcp://10.0.3.1:10059/
[INFO] |yarp.os.impl.PortCoreOutputUnit|/wholeBodyDynamics/left_arm/rpc:o| Sending output from /wholeBodyDynamics/left_arm/rpc:o to /ergocubSim/left_arm/rpc:i using tcp
[INFO] |yarp.os.impl.PortCoreOutputUnit|/wholeBodyDynamics/left_arm/command:o| Sending output from /wholeBodyDynamics/left_arm/command:o to /ergocubSim/left_arm/command:i using udp
[INFO] |yarp.os.impl.PortCoreInputUnit|/wholeBodyDynamics/left_arm/stateExt:i| Receiving input from /ergocubSim/left_arm/stateExt:o to /wholeBodyDynamics/left_arm/stateExt:i using udp
[INFO] |yarp.dev.PolyDriver|left_arm_mc| Created device <remote_controlboard>. See C++ class RemoteControlBoard for documentation.
[INFO] Opening device right_arm_mc with parameters [("robotName" = "ergoCubGazeboV1"), ("remote" = "/ergocubSim/right_arm"), ("local" = "/wholeBodyDynamics/right_arm")]
[DEBUG] |yarp.dev.PolyDriver|right_arm_mc| Parameters are (device remote_controlboard) (id right_arm_mc) (local "/wholeBodyDynamics/right_arm") (remote "/ergocubSim/right_arm") (robotName ergoCubGazeboV1)
[INFO] |yarp.os.Port|/wholeBodyDynamics/right_arm/rpc:o| Port /wholeBodyDynamics/right_arm/rpc:o active at tcp://10.0.3.1:10060/
[... 109 lines cut ...]
[DEBUG] wholeBodyDynamics Statistics: Opening remapper virtual sensors.
[DEBUG] |yarp.dev.PolyDriver|virtualAnalogRemapper| Parameters are (axesNames (torso_roll torso_pitch torso_yaw l_shoulder_pitch l_shoulder_roll l_shoulder_yaw l_elbow l_wrist_pitch l_wrist_yaw r_shoulder_pitch r_shoulder_roll r_shoulder_yaw r_elbow r_wrist_pitch r_wrist_yaw l_hip_pitch l_hip_roll l_hip_yaw l_knee l_ankle_pitch l_ankle_roll r_hip_pitch r_hip_roll r_hip_yaw r_knee r_ankle_pitch r_ankle_roll)) (device virtualAnalogRemapper)
[INFO] |yarp.dev.PolyDriver|virtualAnalogRemapper| Created device <virtualAnalogRemapper>. See C++ class yarp::dev::VirtualAnalogRemapper for documentation.
[DEBUG] wholeBodyDynamics Statistics: Remapper virtual sensors opened in  0.00642109 s.
[DEBUG] wholeBodyDynamics Statistics: Opening contact frames.
[WARNING] subModel no  0  has the maximum number of variables (6). The frame  r_hand_palm  will not be added to it.
[WARNING] subModel no  0  has the maximum number of variables (6). The frame  root_link  will not be added to it.
[WARNING] subModel no  0  has the maximum number of variables (6). The frame  l_foot_front  will not be added to it.
[WARNING] subModel no  0  has the maximum number of variables (6). The frame  l_foot_rear  will not be added to it.
[WARNING] subModel no  0  has the maximum number of variables (6). The frame  r_foot_front  will not be added to it.
[WARNING] subModel no  0  has the maximum number of variables (6). The frame  r_foot_rear  will not be added to it.
[... cut ...]
[ERROR] wholeBodyDynamics: Problem in opening external wrenches port.
[ERROR] |yarp.dev.PolyDriver|wholebodydynamics| Driver <wholebodydynamics> was found but could not open
[WARNING] Cannot open device wholebodydynamics
[WARNING] Cannot open device wholebodydynamics
[WARNING] There was some problem opening one or more devices. Please check the log and your configuration
[ERROR] One or more devices failed opening... see previous log messages for more info
[INFO] Closing device wholebodydynamics
```

traversaro, 2024-05-02:
Thanks for reporting the issue.

traversaro, 2024-05-02:
@SimoneMic which models are you launching? I tried to replicate this, but I could not start any ergocub models as it was making gazebo crash.

SimoneMic, 2024-05-02:
Hi @traversaro I tried it on ergoCubGazeboV1 and 1_1.

Now it works with 0.10.0!

traversaro, 2024-05-02:
> Hi @traversaro I tried it on ergoCubGazeboV1 and 1_1.
> 
> Now it works with 0.10.0!

False alarm, I had an old version of gazebo-yarp-plugins. If it works with 0.10.0. then probably there is a regression in 0.11.0 .

traversaro, 2024-05-02:
The bug is in https://github.com/robotology/whole-body-estimators/pull/188/files#diff-938e8a6e8fc4690975a0955e2f3a91a8a23c313180dda2dea0fc6cbea41046d2R386 . Basically, we switched to use the `` method, but by doing so we are not executing anymore https://github.com/robotology/idyntree/blob/42e779bbecb98ab934450afee04d100c6274cee7/src/estimation/src/ExtWrenchesAndJointTorquesEstimator.cpp#L146-L160, i.e. we are also removing the FT sensors fixed joint from the model, and then the loading fails as the FT sensors are not there.

traversaro, 2024-05-02:
@SimoneMic the issue should have been fixed in the latest master of whole-body-estimators, and releases as v0.11.1 . I will re-open the issue, if you can confirm that the issue has been fixed, I guess we can close it.

SimoneMic, 2024-05-02:
Yes we can close this, thanks a lot!

closed 2024-05-02 by traversaro