mesh-iit/ergocub-software #248: Failed to compile on Windows on fresh conda environment

S-Dafarra, 2024-05-30:
I am having the following error
```
  Performing configure step for 'ergocub-software'
  Not searching for unused variables given on the command line.
  loading initial cache file D:/robotology-superbuild/build/src/ergocub-software/CMakeFiles/YCMTmp/ergocub-software-cac
  he-Release.cmake
  -- Found YCM: D:/robotology-superbuild/build/install/share/cmake/YCM (found version "0.16.2")
  -- Found YARP: D:/robotology-superbuild/build/install/lib/cmake/YARP (found version "3.9.0")
  -- Found YCM: D:/robotology-superbuild/build/install/share/cmake/YCM (found version "0.16.2")
  CMake Error at D:/miniforge/envs/robsub/Library/share/cmake-3.28/Modules/FindPackageHandleStandardArgs.cmake:230 (mes
  sage):
  -- Configuring incomplete, errors occurred!
    Could NOT find OpenCV (missing: OpenCV_CONFIG) (Required is at least
    version "4.2.0")
  Call Stack (most recent call first):
    D:/miniforge/envs/robsub/Library/share/cmake-3.28/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_M
  ESSAGE)
    D:/robotology-superbuild/build/install/share/YCM/find-modules/FindOpenCV.cmake:99 (find_package_handle_standard_arg
  s)
    src/modules/ergoCubEmotions/CMakeLists.txt:10 (find_package)

C:\Program Files\Microsoft Visual Studio\2022\Enterprise\MSBuild\Microsoft\VC\v170\Microsoft.CppCommon.targets(254,5):
error MSB8066: Custom build for 'D:\robotology-superbuild\build\CMakeFiles\02e52bd68bfde91f027028f928af572a\ergocub-sof
tware-configure.rule;D:\robotology-superbuild\build\CMakeFiles\02e52bd68bfde91f027028f928af572a\ergocub-software-build.
rule;D:\robotology-superbuild\build\CMakeFiles\02e52bd68bfde91f027028f928af572a\ergocub-software-install.rule;D:\roboto
logy-superbuild\build\CMakeFiles\0a2a271fa810df30c986e1c85f4af500\ergocub-software-complete.rule;D:\robotology-superbui
ld\build\CMakeFiles\c83b3992172f690c7dfa6d73e27a7d25\ergocub-software.rule' exited with code 1. [D:\robotology-superbui
ld\build\ergocub-software.vcxproj]
```
The version of ``ergocub-software`` is 
```
commit e86bde7e2467c776e119761ac89814d0d2dfc3de (HEAD -> master, tag: v0.7.4, origin/master, origin/HEAD)
Author: Nicolò Genesio (二コゲネ) <nicolo.genesio@iit.it>
Date:   Mon May 27 16:17:47 2024 +0200

    Update CHANGELOG.md
```

<details>
<summary>Here is the <code>conda list </code> </summary>

```
(robsub) D:\robotology-superbuild\src\ergocub-software>conda list
# packages in environment at D:\miniforge\envs\robsub:
#
# Name                    Version                   Build  Channel
ace                       8.0.0                he0c23c2_0    conda-forge
aiohttp                   3.9.5           py312he70551f_0    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
aom                       3.8.2                h63175ca_0    conda-forge
asio                      1.29.0               h63175ca_0    conda-forge
assimp                    5.4.1                h0dbab56_0    conda-forge
asttokens                 2.4.1              pyhd8ed1ab_0    conda-forge
attrs                     23.2.0             pyh71513ae_0    conda-forge
blosc                     1.21.5               hdccc3a2_0    conda-forge
boost                     1.84.0               h7e22eef_3    conda-forge
brotli                    1.0.9                hcfcfb64_9    conda-forge
brotli-bin                1.0.9                hcfcfb64_9    conda-forge
bzip2                     1.0.8                hcfcfb64_5    conda-forge
c-compiler                1.7.0                hcfcfb64_1    conda-forge
ca-certificates           2024.2.2             h56e8100_0    conda-forge
cached-property           1.5.2                hd8ed1ab_1    conda-forge
cached_property           1.5.2              pyha770c72_1    conda-forge
cairo                     1.18.0               h1fef639_0    conda-forge
certifi                   2024.2.2           pyhd8ed1ab_0    conda-forge
clangdev                  5.0.0                   flang_3    conda-forge
cmake                     3.28.3               hf0feee3_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
compilers                 1.7.0                h57928b3_1    conda-forge
contourpy                 1.2.1           py312h0d7def4_0    conda-forge
[... 239 lines cut ...]
wslink                    2.0.4              pyhd8ed1ab_0    conda-forge
x264                      1!164.3095           h8ffe710_2    conda-forge
x265                      3.5                  h2d74725_3    conda-forge
xorg-kbproto              1.0.7             hcd874cb_1002    conda-forge
xorg-libice               1.1.1                hcd874cb_0    conda-forge
xorg-libsm                1.2.4                hcd874cb_0    conda-forge
xorg-libx11               1.8.9                hefa74cf_0    conda-forge
xorg-libxau               1.0.11               hcd874cb_0    conda-forge
xorg-libxdmcp             1.1.3                hcd874cb_0    conda-forge
xorg-libxext              1.3.4                hcd874cb_2    conda-forge
xorg-libxpm               3.5.17               hcd874cb_0    conda-forge
xorg-libxt                1.3.0                hcd874cb_1    conda-forge
xorg-xextproto            7.3.0             hcd874cb_1003    conda-forge
xorg-xproto               7.0.31            hcd874cb_1007    conda-forge
xz                        5.2.6                h8d14728_0    conda-forge
yaml                      0.2.5                h8ffe710_2    conda-forge
yarl                      1.9.4           py312he70551f_0    conda-forge
zeromq                    4.3.5                he1f189c_4    conda-forge
zlib                      1.2.13               h2466b09_6    conda-forge
zstd                      1.5.6                h0ea2cb4_0    conda-forge
```

</details>

cc @traversaro @Nicogene

S-Dafarra, 2024-05-31:
> Sorry @S-Dafarra, can you report the `conda list` of the environment?

It is already at the end of the comment 😉

traversaro, 2024-05-31:
Ah Sorry I missed it, thanks!

Nicogene, 2024-06-03:
@S-Dafarra can you confirm that the issue now has been solved?

traversaro, 2024-06-03:
> > @S-Dafarra can you confirm that the issue now has been solved?
> 
> I cannot before next week unfortunately

By the way, I do not think this will be solved by https://github.com/robotology/ycm-cmake-modules/pull/449 . Anyhow, https://github.com/robotology/ycm-cmake-modules/pull/449 was a fix that was necessaey (I had no idea how things were working before).

S-Dafarra, 2024-06-10:
It seems that it worked. I pulled ``ycm`` master, removed ``ergocub-software`` build, and let the superbuild compile the ``ergocub-software`` target. I did not encounter the issue anymore.

traversaro, 2024-06-10:
Great, thanks!

closed 2024-06-10 by traversaro

traversaro, 2024-06-11:
I wonder if this was related to https://github.com/robotology/robotology-superbuild/issues/1664#issuecomment-2160845262 .