mesh-iit/ergocub-software #257: ergoCubGazeboV1_1: cannot read encoders after resetting world when using `gazebo_yarp_clock`

PasMarra, 2024-08-22:
When running Gazebo with the `gazebo_yarp_clock`, if I reset the world (`ctrl+R`) and, then, I try to run the `yarpmotorgui`, this latter fails to read the encoders.
The problem does not happen when not using `gazebo_yarp_clock`.

## How to recreate the error:

<details><summary>conda list -n metaCub</summary>

```
pasquale@IITICB001LW003:~$ conda list -n metaCub
# packages in environment at /home/pasquale/miniforge3/envs/metaCub:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
ace                       8.0.0                hac33072_1    conda-forge
addict                    2.4.0                    pypi_0    pypi
aiohttp                   3.9.5           py310h2372a71_0    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
alsa-lib                  1.2.12               h4ab18f5_0    conda-forge
ampl-mp                   3.1.0             h2cc385e_1006    conda-forge
aom                       3.8.2                h59595ed_0    conda-forge
assimp                    5.3.1                hfb0e8fe_2    conda-forge
asttokens                 2.4.1              pyhd8ed1ab_0    conda-forge
async-timeout             4.0.3              pyhd8ed1ab_0    conda-forge
atk-1.0                   2.38.0               hd4edc92_1    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
attrs                     23.2.0             pyh71513ae_0    conda-forge
aws-c-auth                0.7.17               he0b1f16_2    conda-forge
aws-c-cal                 0.6.11               heb1d5e4_0    conda-forge
aws-c-common              0.9.15               hd590300_0    conda-forge
aws-c-compression         0.2.18               hce8ee76_3    conda-forge
aws-c-event-stream        0.4.2                h01f5eca_8    conda-forge
aws-c-http                0.8.1               hdb68c23_10    conda-forge
aws-c-io                  0.14.7               hbfbeace_6    conda-forge
aws-c-mqtt                0.10.3               h50844eb_4    conda-forge
aws-c-s3                  0.5.7                hb7bd14b_1    conda-forge
aws-c-sdkutils            0.1.15               hce8ee76_3    conda-forge
aws-checksums             0.1.18               hce8ee76_3    conda-forge
aws-crt-cpp               0.26.6               hf567797_4    conda-forge
aws-sdk-cpp               1.11.267             hbf3e495_6    conda-forge
azure-core-cpp            1.11.1               h91d86a7_1    conda-forge
azure-identity-cpp        1.6.0                hf1915f5_1    conda-forge
azure-storage-blobs-cpp   12.10.0              h00ab1b0_1    conda-forge
azure-storage-common-cpp  12.5.0               h94269e2_4    conda-forge
binutils                  2.40                 h4852527_7    conda-forge
binutils_impl_linux-64    2.40                 ha1999f0_7    conda-forge
binutils_linux-64         2.40                 hb3c18ed_0    conda-forge
bipedal-locomotion-framework 0.18.0               ha770c72_6    conda-forge
bipedal-locomotion-framework-python 0.18.0          py310he3eb0c9_6    conda-forge
blinker                   1.8.2                    pypi_0    pypi
blosc                     1.21.5               h0f2a231_0    conda-forge
boost-cpp                 1.82.0               h44aadfe_6    conda-forge
brotli                    1.1.0                hd590300_1    conda-forge
brotli-bin                1.1.0                hd590300_1    conda-forge
bullet-cpp                3.25                 hcc13569_2    conda-forge
bzip2                     1.0.8                h4bc722e_7    conda-forge
c-ares                    1.32.2               h4bc722e_0    conda-forge
c-compiler                1.7.0                hd590300_1    conda-forge
ca-certificates           2024.7.4             hbcca054_0    conda-forge
cached-property           1.5.2                hd8ed1ab_1    conda-forge
cached_property           1.5.2              pyha770c72_1    conda-forge
cairo                     1.18.0               h3faef2a_0    conda-forge
casadi                    3.6.5           py310he84b9c3_4    conda-forge
certifi                   2024.7.4           pyhd8ed1ab_0    conda-forge
cfitsio                   4.3.1                hbdc6101_0    conda-forge
charset-normalizer        3.3.2                    pypi_0    pypi
click                     8.1.7                    pypi_0    pypi
collada-dom               2.5.0                ha438cbf_7    conda-forge
comm                      0.2.2                    pypi_0    pypi
configargparse            1.7                      pypi_0    pypi
[... 481 lines cut ...]
[WARNING] robotDescriptionServer not found, robot parts will be set manually.
[DEBUG] Appending /ergocubSim/head
[DEBUG] Appending /ergocubSim/torso
[DEBUG] Appending /ergocubSim/left_arm
[DEBUG] Appending /ergocubSim/right_arm
[DEBUG] Appending /ergocubSim/left_leg
[DEBUG] Appending /ergocubSim/right_leg
[DEBUG] Checking the existence of: /yarpmotorgui0//ergocubSim/head/rpc:o 
[DEBUG] ADDRESS is:  
[DEBUG] |yarp.dev.PolyDriver|remote_controlboard| Parameters are (carrier udp) (device remote_controlboard) (local "/yarpmotorgui0//ergocubSim/head") (remote "/ergocubSim/head")
[INFO] |yarp.os.Port|/yarpmotorgui0//ergocubSim/head/rpc:o| Port /yarpmotorgui0//ergocubSim/head/rpc:o active at tcp://10.240.78.138:10052/
[INFO] |yarp.os.Port|/yarpmotorgui0//ergocubSim/head/command:o| Port /yarpmotorgui0//ergocubSim/head/command:o active at tcp://10.240.78.138:10053/
[INFO] |yarp.os.Port|/yarpmotorgui0//ergocubSim/head/stateExt:i| Port /yarpmotorgui0//ergocubSim/head/stateExt:i active at tcp://10.240.78.138:10054/
[INFO] |yarp.os.impl.PortCoreOutputUnit|/yarpmotorgui0//ergocubSim/head/rpc:o| Sending output from /yarpmotorgui0//ergocubSim/head/rpc:o to /ergocubSim/head/rpc:i using tcp
[INFO] |yarp.os.impl.PortCoreOutputUnit|/yarpmotorgui0//ergocubSim/head/command:o| Sending output from /yarpmotorgui0//ergocubSim/head/command:o to /ergocubSim/head/command:i using udp
[INFO] |yarp.os.impl.PortCoreInputUnit|/yarpmotorgui0//ergocubSim/head/stateExt:i| Receiving input from /ergocubSim/head/stateExt:o to /yarpmotorgui0//ergocubSim/head/stateExt:i using udp
[INFO] |yarp.dev.PolyDriver|remote_controlboard| Created device <remote_controlboard>. See C++ class RemoteControlBoard for documentation.
[DEBUG] Setting a valid finder 
[INFO] |yarp.os.Port|/yarpmotorgui/head/sequence:o| Port /yarpmotorgui/head/sequence:o active at tcp://10.240.78.138:10055/
[DEBUG] Initializing interfaces...
[DEBUG] Opening interfaces...
[ERROR] head iencs->getEncoders() failed, retrying...
[ERROR] head iencs->getEncoders() failed, retrying...
[ERROR] head iencs->getEncoders() failed, retrying...
```

Nicogene, 2024-08-23:
Hi @PasMarra !

I think that this is more a `gazebo-yarp-plugins` issue, or it is happening only with the `ergoCubGazeboV1_1` model?

PasMarra, 2024-08-26:
> Hi @PasMarra !
> 
> I think that this is more a `gazebo-yarp-plugins` issue, or it is happening only with the `ergoCubGazeboV1_1` model?

Yes, you're right. Actually it happens also for `ergoCubGazeboV1_2`.

PasMarra, 2024-08-26:
Closing as the problem has been reported [here](https://github.com/robotology/gazebo-yarp-plugins/issues/685#issue-2486219516)

closed 2024-08-26 by PasMarra