mesh-iit/ergocub-software #260: Possible wrong position of waist IMU ergoCubSN001

fils99, 2024-09-11:
## Task description

Today, during experiments with ergoCubSN001, we wanted to check the position and orientation of the waist IMU. First of all we looked to the IMU in the real robot. Afterwards, we compared them with the ones of the URDF model (we use this version of `ergocub-software` e4b567e67e746340051e296b0d9ed74c164f454b). 

[image]

As we can see in the image, the position of waist IMU in the real robot seems a bit different wrt the URDF one.

## Definition of Done

Understand if position of waist IMU is correct

cc @LoreMoretti @GiulioRomualdi

GiulioRomualdi, 2024-09-11:
cc @Nicogene and @traversaro

Nicogene, 2024-09-11:
The imu in the waist has been added in the SN000 urdf as XML blob (since the simulation model cad is unmaintained):
- https://github.com/icub-tech-iit/ergocub-software/pull/99

I didn't recall that is still exported as XML blob also in newer models:

https://github.com/icub-tech-iit/ergocub-software/blob/b030a601111490a1e0739d14ff45cee5d1181555/urdf/creo2urdf/data/ergocub1_1/base_1_1.yaml#L1142-L1150

Maybe @Mick3Lozzo recall why we didn't exported that frame, but we should do it from the cad (cc @fiorisi @Lawproto)

fiorisi, 2024-09-11:
If I correctly understood the problem I think that the IMU was moved because of a possible collision with the torso covers: https://github.com/icub-tech-iit/project-ergocub/issues/107#issuecomment-1531510392.

Nicogene, 2024-09-12:
> If I correctly understood the problem I think that the IMU was moved because of a possible collision with the torso covers: [icub-tech-iit/project-ergocub#107 (comment)](https://github.com/icub-tech-iit/project-ergocub/issues/107#issuecomment-1531510392).

Good catch!

Nicogene, 2024-10-02:
Today I checked the transform between the `SCSYS_TORSO_1` and the waist IMU csys and here is the result

[image]

I used this snippet using the idyntree binding for getting `xyzrpy`:
``` import idyntree

rpy = idyntree.bindings.Rotation(0, 0, -1, -1, 0.000105208, 0, 0.000105208, 1, 0).asRPY()
```

And I got:

```diff
-<origin xyz="0.0354497 0.00639688 0.060600" rpy="1.5708 0 -1.5708"/> 
+<origin xyz="0.0606 -0.00639428 0.01745" rpy="1.5708 -0.000105208 -1.5708"/> 
```

The difference seems not only a displacement, I used the correct CSYS for computing the transform? Are those csys correct? I took them from `ecub.asm` (cc @fiorisi @Mick3Lozzo)

Nicogene, 2024-10-02:
xref:
- https://github.com/icub-tech-iit/ergocub-software/issues/215

Mick3Lozzo, 2024-10-02:
> xref:
> 
> * [ergoCub SN001/ ergoCubGazeboV1_1*: the waist imu is still added as xml blob to the urdf #215](https://github.com/icub-tech-iit/ergocub-software/issues/215)

@Nicogene what kind of IMU support is AMI using? I am going to check.

cc @fiorisi

Nicogene, 2024-10-02:
> [[image]@Nicogene](https://github.com/Nicogene) what kind of IMU support is AMI using? I am going to check.
>
> cc [[image]@fiorisi](https://github.com/fiorisi)

The xsens mti that is mounted in the torso, I got that transform pulling the latest `cad-mechanics`

Mick3Lozzo, 2024-10-02:
> The difference seems not only a displacement, I used the correct CSYS for computing the transform. Are those csys correct? I took them from ecub.asm (cc @fiorisi @Mick3Lozzo)

[image]

These are the same values I found @Nicogene.

Nicogene, 2024-10-02:
Thanks @Mick3Lozzo !
That very small rotation on pitch is strange ( we can also approximate it to zero ), but it this is the correct pose I will change the urdf accordingly

Nicogene, 2024-10-08:
Something has to be fixed:
- https://github.com/icub-tech-iit/ergocub-software/pull/267#issuecomment-2399838233

Probably on creo2urdf side

Nicogene, 2024-10-09:
Hi @fils99, with:

- #267 

The position of `waist_imu_0` is:

[image]

That seems coherent with

[image]

closed 2024-10-09 by Nicogene

fils99, 2024-10-09:
Thanks @Nicogene!

PR #267 (2024-10-09): Fix the position of the imu in the waist
It fixes #260
commit c4a1fb16 (2024-10-09): Fix the position of the imu in the waist

It fixes #260