mesh-iit/ergocub-software #263: Integrate nice meshes manually crafted into the repo-pipeline

Nicogene, 2024-09-27:
## Task description

This task is probably cross-repository with `creo2urdf`, @supermonkey-design did an amazing job on remeshing the the existing URDF mesh, and after a chat w/ @fiorisi @traversaro @GiulioRomualdi it came out that it would be nice to add them to this repo, alongside the "ugly" ones.

In case these meshes are heavy for the simulation we might think to change creo2urdf in order to make `assignedCollisionGeometry` handle also complex meshes, and then use the ugly ones for the collisions, keeping the nice one as visual

## Definition of Done

We integrated the @supermonkey-design  meshes

Nicogene, 2024-10-02:
As first step I simply added all the new meshes to the repo and find/replace `.stl` with `.obj` in the urdf.

I am getting this warning on gazebo side:
```
[Wrn] [Event.cc:61] Warning: Deleting a connection right after creation. Make sure to save the ConnectionPtr from a Connect call
[Wrn] [OBJLoader.cc:79] Material file [ ./sim_ecub_1-1_l_hip_1.obj.mtl ] not found in a path : /home/ngenesio/robotology/robotology-superbuild/build/install/share/ergoCub/meshes/simmechanics/
Failed to load material file(s). Use default material.
material [ 'material_0' ] not found in .mtl

[Wrn] [OBJLoader.cc:146] Missing material for shape[] in OBJ file[/home/ngenesio/robotology/robotology-superbuild/build/install/share/ergoCub/meshes/simmechanics/sim_ecub_l_hip_1.obj]
[Wrn] [OBJLoader.cc:79] Material file [ ./sim_ecub_1-1_r_hip_1.obj.mtl ] not found in a path : /home/ngenesio/robotology/robotology-superbuild/build/install/share/ergoCub/meshes/simmechanics/
Failed to load material file(s). Use default material.
material [ 'material_0' ] not found in .mtl

[Wrn] [OBJLoader.cc:146] Missing material for shape[] in OBJ file[/home/ngenesio/robotology/robotology-superbuild/build/install/share/ergoCub/meshes/simmechanics/sim_ecub_r_hip_1.obj]
[Wrn] [OBJLoader.cc:79] Material file [ ./sim_ecub_1-1_l_shoulder_1.obj.mtl ] not found in a path : /home/ngenesio/robotology/robotology-superbuild/build/install/share/ergoCub/meshes/simmechanics/
Failed to load material file(s). Use default material.
material [ 'material_0' ] not found in .mtl

[Wrn] [OBJLoader.cc:146] Missing material for shape[] in OBJ file[/home/ngenesio/robotology/robotology-superbuild/build/install/share/ergoCub/meshes/simmechanics/sim_ecub_l_shoulder_1.obj]
[Wrn] [OBJLoader.cc:79] Material file [ ./sim_ecub_1-1_head_neck_3.obj.mtl ] not found in a path : /home/ngenesio/robotology/robotology-superbuild/build/install/share/ergoCub/meshes/simmechanics/
Failed to load material file(s). Use default material.
material [ 'material_0' ] not found in .mtl

[Wrn] [OBJLoader.cc:146] Missing material for shape[] in OBJ file[/home/ngenesio/robotology/robotology-superbuild/build/install/share/ergoCub/meshes/simmechanics/sim_ecub_head_neck_3.obj]
[Wrn] [OBJLoader.cc:79] Material file [ ./sim_ecub_1-1_r_shoulder_1.obj.mtl ] not found in a path : /home/ngenesio/robotology/robotology-superbuild/build/install/share/ergoCub/meshes/simmechanics/
Failed to load material file(s). Use default material.
material [ 'material_0' ] not found in .mtl

[Wrn] [OBJLoader.cc:146] Missing material for shape[] in OBJ file[/home/ngenesio/robotology/robotology-superbuild/build/install/share/ergoCub/meshes/simmechanics/sim_ecub_r_shoulder_1.obj]
```

and unfortunately the meshes seems to collapse:

[image]

cc @Supermonkey-design

Supermonkey-design, 2024-10-02:
> I am getting this warning on gazebo side:

Sorry @Nicogene, I've forgotten to change some of the mtl references. It should be ok now.
For the collapsing, I don't know. To launch gazebo or idyntree, in the urdf I've only ever used the path for the visuals (obj) and left the rest as it was.

Nicogene, 2024-10-02:
> > I am getting this warning on gazebo side:
> 
> Sorry [[image]@Nicogene](https://github.com/Nicogene), I've forgotten to change some of the mtl references. It should be ok now. 

The warning are now disappeared thanks!

For the pose of the meshes instead, with stl usually happens when we export them wrt the part csys (that usually is not where is the link csys), but for these obj I don't know how the procedure work.

>To launch gazebo or idyntree, in the urdf I've only ever used the path for the visuals (obj) and left the rest as it was.

It is strange indeed that I am experiencing this problem, I just change the paths too 🤔

Nicogene, 2024-10-02:
The curious thing is that with the 1.1 meshes this does not happen:
[image]

> Did you replace just collision ao also visuals?

Yes it should!

traversaro, 2024-10-02:
Is anything printed if you run gazebo --verbose ?

Nicogene, 2024-10-02:
I found out what the problem is.
**Preface:** The `SN001/V1` models have never been created via `creo2urdf`, they have been built via the old procedure:
- https://github.com/icub-tech-iit/ergocub-software/issues/106
- https://github.com/icub-tech-iit/ergocub-software/tree/master/urdf/creo2urdf/data/ergocub1_0

Comparing side-by-side the `V1` and `V1_1` urdf I see an important different on how the meshes are handled:
## V_1
```urdf
    <visual>
      <origin xyz="0 0 0.1785" rpy="0 0 1.5707963267948966"/>
      <geometry>
        <mesh filename="package://ergoCub/meshes/simmechanics/sim_ecub_root_link.obj" scale="0.001 0.001 0.001"/>
      </geometry>
      <material name="root_link_color">
        <color rgba="0.9 0.9 0.9 1"/>
      </material>
    </visual>
    <collision>
      <origin xyz="0 0 0.1785" rpy="0 0 1.5707963267948966"/>
      <geometry>
        <mesh filename="package://ergoCub/meshes/simmechanics/sim_ecub_root_link.obj" scale="0.001 0.001 0.001"/>
      </geometry>
    </collision>
```

## V1_1
```urdf
    <visual>
      <origin xyz="0 0 0" rpy="0 -0 0"/>
      <geometry>
        <mesh filename="package://ergoCub/meshes/simmechanics/sim_ecub_1-1_root_link.obj" scale="0.001 0.001 0.001"/>
      </geometry>
      <material name="">
        <color rgba="0.6 0.6 0.6 1"/>
      </material>
    </visual>
    <collision>
      <origin xyz="0 0 0" rpy="0 -0 0"/>
      <geometry>
        <mesh filename="package://ergoCub/meshes/simmechanics/sim_ecub_1-1_root_link.obj" scale="0.001 0.001 0.001"/>
      </geometry>
```

Long story short, using the old pipeline we have meshes exported w.r.t the csys and then the pose is adjusted using the `origin` tag, with the new procedure instead we directly export the meshes respect the correct csys, then the `origin` tag is simply an identity.

@Supermonkey-design just changed the names of the meshes and changed the color starting from the V1_1 ones, probably the V1 have to be export respect `CSYS` or as last resort I can change all the origin tags in 000 000 also in V1 (

GiulioRomualdi, 2024-10-02:
I was wondering if it makes sense to also update the meshes of the sn000 since it is created with a machinery not supported anymore

Nicogene, 2024-10-03:
> I was wondering if it makes sense to also update the meshes of the sn000 since it is created with a machinery not supported anymore

You are right, my only fear is that if we want to roll-back to the "ugly" meshes, if we change the urdf the stl meshes have the wrong placement. I quickly tested the impact on the rtf and seems that these new meshes are quite light, but I did not do "complex" task, such as walking grasping etc

Nicogene, 2024-10-03:
Instead about the weight of these meshes, I don't know if we might commit the with git lfs or not, until now for using `ergocub-software` git lfs was not needed and introducing can be considered a breaking change? @traversaro @pattacini

pattacini, 2024-10-03:
Hi @Nicogene 

The main problem with Git LFS is when we aim to apply it to extensions already hosted in the repo.
Do you know if this is the case here? Do these new meshes have new file extensions that are currently not hosted?

pattacini, 2024-10-03:
Also, what file size we're talking about?

traversaro, 2024-10-03:
> I've made a little definition in Grasshopper to remap the position of V1_1 meshes into the frame of V1 meshes (so without exporting everything again from creo).

How is this process done? I am always a bit afraid of process that rely on visually checking the aligned of meshes. Did the kinematics changed between ergoCubSN000, 1, 2? If not, we can compute the correct placement from the kinematics, without trying to align the meshes.

Nicogene, 2024-10-03:
> How is this process done? I am always a bit afraid of process that rely on visually checking the aligned of meshes. Did the kinematics changed between ergoCubSN000, 1, 2? If not, we can compute the correct placement from the kinematics, without trying to align the meshes.

A part of the fingers they should be kinematically identical

> Also, what file size we're talking about?

@pattacini we are passing from **43.8 MB** to **96.8 MB**, this means we are doubling the size, but still seems a reasonable size

pattacini, 2024-10-03:
>I would apply in case only to .obj and .obj.mtl that have never committed before

Good 👍🏻 

>@pattacini we are passing from 43.8 MB to 96.8 MB, this means we are doubling the size, but still seems a reasonable size

Not really, we're approaching the GH limit of 100 MB per file, hence LFS seems necessary.

Supermonkey-design, 2024-10-03:
> How is this process done? I am always a bit afraid of process that rely on visually checking the aligned of meshes.

Me too, honestly, in this case. My idea was also simple: if the meshes are identical, I can read the shrinkwrapped mesh from SN001 and SN000, compute their bounding box centroids, create a vector from these two points, and use it to remap the position of the detailed meshes.
But for example I had to match the rotation of the bound box of the arms and shoulder. I can do a better job and make it automatic by calculating the frame, but as there were only three meshes, I did it manually.

traversaro, 2024-10-03:
Yes, that is exactly what I was afraid of. If the meshes are indeed identical, perhaps to use the V1_1 meshes in V1 we can simply set the `origin` tag of `V1` to `<origin xyz="0 0 0" rpy="0 0 0"/>`, instead of moving the meshes around?

Nicogene, 2024-10-03:
For curiosity I just exported the SN000 urdf using `creo2urdf` and at first glance seems a well-formed urdf

Maybe this is solved:
- https://github.com/icub-tech-iit/ergocub-software/issues/177

Because in these months `creo2urdf` became more robust, my only doubt is if we are leaving some hack we made by hand behind.
In all these months I tried to keep the yamls updated even if not used

traversaro, 2024-10-03:
Note that this may change the formatting of the existing xml if `tree.write` uses another formatting w.r.t. to the one currently used.

Nicogene, 2024-10-03:
@Supermonkey-design Here:
- https://github.com/icub-tech-iit/ergocub-software/pull/266

I put all the origin to 000, I can take the old meshes and see if now they work

Nicogene, 2024-10-04:
> Yes, that is exactly what I was afraid of. If the meshes are indeed identical, perhaps to use the V1_1 meshes in V1 we can simply set the `origin` tag of `V1` to `<origin xyz="0 0 0" rpy="0 0 0"/>`, instead of moving the meshes around?

Unfortunately the meshes for the hands are different, so right now the V1 hands looks like this:

[image]

[image]

This is due to the fact that the URDF contains the V1 transform between links, but the meshes are the ones of the V1_1, that are longer and have some problems of frame definitions:

- https://github.com/icub-tech-iit/ergocub-software/issues/199#issuecomment-1862611713

"This is not creating problems, just to be aware of it" 

Naif me of 2023 :D 

So with @Supermonkey-design we decided to:

- Take the stl of V1 obtained via creo2urdf
- Convert them to obj

That's it?

Not really, the simulation model of V1 has yet another problem:

- https://github.com/icub-tech-iit/ergocub-software/issues/82

That was fixed by @xEnVrE by removing the leverism by hand and committing the stl directly.

Nicogene, 2024-10-04:
> Good 👍🏻
>
>
> Not really, we're approaching the GH limit of 100 MB per file, hence LFS seems necessary.

About this point maybe I was not clear, but **96.8 MB is the total** of the entire `ergocub-software` folder, the 100 MB limit seems to be related to a single file:
- https://docs.github.com/en/repositories/working-with-files/managing-large-files/about-large-files-on-github#about-size-limits-on-github

The limit on the repo seems to be 1 GB.

I would rather avoid using `git lfs` because this may lead to problems for users, and we should specify to install git lfs both here and in the robotology-superbuild documentation since `ergocub-software` would be the first repo that requires it.

PR #266 (2024-10-07): Add the nice meshes for ergocub V1/V1_1/V1_2
This PR adds the meshes made by @Supermonkey-design that visually improve the ergocub models without impaction on the RTF

| **SN000** |   **SN001/2** |
|----------|:---------:|
| [image] |  [image] |

It fixes #263