mesh-iit/ergocub-software #265: Document/simply the usage of ergocub models in mujoco

traversaro, 2024-10-02:
A bit of spinoff of https://github.com/robotology/icub-models/issues/155, but as most people in AMI now use ergocub  we want to focus on ergocub here.

As first step, we may need to to understand how people are currently loading the ergocub models in mujoco, can you share this @diegoferigo @CarlottaSartore @vpunithreddy @Giulero @giotherobot @xela-95 @flferretti @GiulioRomualdi , also via pointers to existing code? With this, info we can understand possible outcomes:
* Document how to load existing ergocub-software models using some library in mujoco
* Add some automatically generated output models to ergocub-software.

traversaro, 2024-10-02:
xref: https://github.com/google-deepmind/mujoco/issues/1432 .

traversaro, 2024-10-02:
> This is my library: https://github.com/ami-iit/mujoco-urdf-loader

So probably a good first pointer is to point ergocub users to https://github.com/ami-iit/mujoco-urdf-loader/blob/main/examples/generate_ergoCub_mjcf.py, not sure if this is enough for what @GiulioRomualdi and @giulero had in mind.

diegoferigo, 2024-10-03:
Inside [ami-iit/jaxsim](https://github.com/ami-iit/jaxsim) we have a [`jaxsim.mujoco`](https://github.com/ami-iit/jaxsim/tree/main/src/jaxsim/mujoco) Python package for visualizing URDF/SDF models.

Refer to the following PRs showing the main features supported by our visualizer:

- https://github.com/ami-iit/jaxsim/pull/83
- https://github.com/ami-iit/jaxsim/pull/125
- https://github.com/ami-iit/jaxsim/pull/202
- https://github.com/ami-iit/jaxsim/pull/133

In particular, the helper for positioning cameras in both the interactive visualizer and the recorder is pretty useful in practice. Also, the support of uneven terrain is nice to have for many applications. 

The Python package has been developed without direct JaxSim dependencies to simplify extracting it to a standalone project in the future -- if neeed.

traversaro, 2024-10-03:
@diegoferigo thanks! Do you have some snippet somewhere (tutorial or test) that just loads a model specified via package:// and simulate it?

diegoferigo, 2024-10-03:
- The combination of a simple simulation with the recorder can be found in [this notebook](https://github.com/ami-iit/jaxsim/blob/main/examples/PD_controller.ipynb).
- For the interactive viewer, instead, a good starting point is the [command line script](https://github.com/ami-iit/jaxsim/blob/0c78499531e149724d255237f40d759a9cd0fa5a/src/jaxsim/mujoco/__main__.py#L164-L186).