mesh-iit/ergocub-software #272: No `mesh.obj` found for the new "nice" meshes when loading gazebo models.

mebbaid, 2024-10-09:
Looking at https://github.com/icub-tech-iit/ergocub-software/pull/266 I wanted to give it a try. However when loading the model in gazebo i get the following 

```bash
[Wrn] [FuelModelDatabase.cc:313] URI not supported by Fuel [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_root_link.obj]
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_root_link.obj]
[Err] [Visual.cc:2956] No mesh specified
[Wrn] [FuelModelDatabase.cc:313] URI not supported by Fuel [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_l_hip_1.obj]
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_l_hip_1.obj]
[Err] [Visual.cc:2956] No mesh specified
[Wrn] [FuelModelDatabase.cc:313] URI not supported by Fuel [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_l_hip_2.obj]
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_l_hip_2.obj]
[Err] [Visual.cc:2956] No mesh specified
[Wrn] [FuelModelDatabase.cc:313] URI not supported by Fuel [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_l_hip_3.obj]
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_l_hip_3.obj]
[Err] [Visual.cc:2956] No mesh specified
[Wrn] [FuelModelDatabase.cc:313] URI not supported by Fuel [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_l_upper_leg.obj]
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_l_upper_leg.obj]
[Err] [Visual.cc:2956] No mesh specified
[Wrn] [FuelModelDatabase.cc:313] URI not supported by Fuel [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_l_lower_leg.obj]
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_l_lower_leg.obj]
[Err] [Visual.cc:2956] No mesh specified
[Wrn] [FuelModelDatabase.cc:313] URI not supported by Fuel [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_l_ankle_1.obj]
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_l_ankle_1.obj]
[Err] [Visual.cc:2956] No mesh specified
[Wrn] [FuelModelDatabase.cc:313] URI not supported by Fuel [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_l_ankle_2.obj]
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_l_ankle_2.obj]
[Err] [Visual.cc:2956] No mesh specified
[Wrn] [FuelModelDatabase.cc:313] URI not supported by Fuel [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_l_foot_front.obj]
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_l_foot_front.obj]
[Err] [Visual.cc:2956] No mesh specified
[Wrn] [FuelModelDatabase.cc:313] URI not supported by Fuel [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_l_foot_rear.obj]
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_l_foot_rear.obj]
[Err] [Visual.cc:2956] No mesh specified
[Wrn] [FuelModelDatabase.cc:313] URI not supported by Fuel [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_r_hip_1.obj]
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_r_hip_1.obj]
[Err] [Visual.cc:2956] No mesh specified
[Wrn] [FuelModelDatabase.cc:313] URI not supported by Fuel [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_r_hip_2.obj]
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_r_hip_2.obj]
[Err] [Visual.cc:2956] No mesh specified
[Wrn] [FuelModelDatabase.cc:313] URI not supported by Fuel [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_r_hip_3.obj]
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_r_hip_3.obj]
[Err] [Visual.cc:2956] No mesh specified
[Wrn] [FuelModelDatabase.cc:313] URI not supported by Fuel [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_r_upper_leg.obj]
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_r_upper_leg.obj]
[Err] [Visual.cc:2956] No mesh specified
[Wrn] [FuelModelDatabase.cc:313] URI not supported by Fuel [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_r_lower_leg.obj]
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_r_lower_leg.obj]
[Err] [Visual.cc:2956] No mesh specified
[Wrn] [FuelModelDatabase.cc:313] URI not supported by Fuel [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_r_ankle_1.obj]
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_r_ankle_1.obj]
[Err] [Visual.cc:2956] No mesh specified
[Wrn] [FuelModelDatabase.cc:313] URI not supported by Fuel [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_r_ankle_2.obj]
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_r_ankle_2.obj]
[Err] [Visual.cc:2956] No mesh specified
[Wrn] [FuelModelDatabase.cc:313] URI not supported by Fuel [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_r_foot_front.obj]
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_r_foot_front.obj]
[Err] [Visual.cc:2956] No mesh specified
[Wrn] [FuelModelDatabase.cc:313] URI not supported by Fuel [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_r_foot_rear.obj]
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_r_foot_rear.obj]
[Err] [Visual.cc:2956] No mesh specified
[Wrn] [FuelModelDatabase.cc:313] URI not supported by Fuel [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_torso_1.obj]
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_torso_1.obj]
[Err] [Visual.cc:2956] No mesh specified
[Wrn] [FuelModelDatabase.cc:313] URI not supported by Fuel [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_torso_2.obj]
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_torso_2.obj]
[Err] [Visual.cc:2956] No mesh specified
[Wrn] [FuelModelDatabase.cc:313] URI not supported by Fuel [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_chest.obj]
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_chest.obj]
[Err] [Visual.cc:2956] No mesh specified
[Wrn] [FuelModelDatabase.cc:313] URI not supported by Fuel [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_l_shoulder_1.obj]
[... 110 lines cut ...]
[Wrn] [FuelModelDatabase.cc:313] URI not supported by Fuel [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_r_hand_pinkie_1.obj]
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_r_hand_pinkie_1.obj]
[Err] [Visual.cc:2956] No mesh specified
[Wrn] [FuelModelDatabase.cc:313] URI not supported by Fuel [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_r_hand_pinkie_2.obj]
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_r_hand_pinkie_2.obj]
[Err] [Visual.cc:2956] No mesh specified
[Wrn] [FuelModelDatabase.cc:313] URI not supported by Fuel [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_r_hand_ring_1.obj]
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_r_hand_ring_1.obj]
[Err] [Visual.cc:2956] No mesh specified
[Wrn] [FuelModelDatabase.cc:313] URI not supported by Fuel [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_r_hand_ring_2.obj]
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_r_hand_ring_2.obj]
[Err] [Visual.cc:2956] No mesh specified
[Wrn] [FuelModelDatabase.cc:313] URI not supported by Fuel [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_r_hand_thumb_1.obj]
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_r_hand_thumb_1.obj]
[Err] [Visual.cc:2956] No mesh specified
[Wrn] [FuelModelDatabase.cc:313] URI not supported by Fuel [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_r_hand_thumb_2.obj]
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_r_hand_thumb_2.obj]
[Err] [Visual.cc:2956] No mesh specified
[... cut ...] 3.obj]
[Wrn] [SystemPaths.cc:459] File or path does not exist [""] [model://ergoCub/meshes/simmechanics/sim_ecub_1-1_r_hand_thumb_3.obj]
[Err] [Visual.cc:2956] No mesh specified
^C[Wrn] [Event.cc:61] Warning: Deleting a connection right after creation. Make sure to save the ConnectionPtr from a Connect call
```

inspecting the PR, indeed the mesh obj files are not there, only the stl and the obj.mtl. Not sure if it is intended and they should be somehow generated by the user or if it is an issue

Nicogene, 2024-10-09:
I think it is .gitignore fault, I will commit also the obj

closed 2024-10-09 by Nicogene

pattacini, 2024-10-09:
What about the Gut LFS layer?

Nicogene, 2024-10-10:
In the end we decided to not involve `git lfs` since the size of the single files are far from the Github limit:
- https://github.com/icub-tech-iit/ergocub-software/issues/263#issuecomment-2393686664

pattacini, 2024-10-10:
Gotcha!

commit 19ad9aeb (2024-10-09): Add .obj meshes

It fixes #272