mesh-iit/ergocub-software #281: Missing gz-sim-yarp plugins fro ergoCubGazeboV1

LoreMoretti, 2025-01-30:
I am trying to load `ergoCubGazeboV1` in `gz-sim`, but it seems that `gz-sim-yarp-plugins` are not added in its `urdf`.

Nicogene, 2025-05-05:
Sorry for the super late reply, I quickly checked and you are right, we lost them at some point, looking the commits of @xela-95 here:
- https://github.com/icub-tech-iit/ergocub-software/pull/230

It seems that they have been added and removed, in fact those urdf are missing in the diff of the PR, @xela-95 do you remember why is that?
Maybe for this problem (only related to the IMU):
- https://github.com/icub-tech-iit/ergocub-software/issues/234

In fact in this:
- https://github.com/icub-tech-iit/ergocub-software/issues/106

We reported that the missing gz-sim support is actually a known limitation, but I don't recall why

xela-95, 2025-05-06:
> Sorry for the super late reply, I quickly checked and you are right, we lost them at some point, looking the commits of [@xela-95](https://github.com/xela-95) here:
> 
> * [Add Gazebo Sim support #230](https://github.com/icub-tech-iit/ergocub-software/pull/230)
> 
> It seems that they have been added and removed, in fact those urdf are missing in the diff of the PR, [@xela-95](https://github.com/xela-95) do you remember why is that? Maybe for this problem (only related to the IMU):
> 
> * [Mismatch between ergoCub 1_0 and 1_1 IMU sensor parent Links #234](https://github.com/icub-tech-iit/ergocub-software/issues/234)
> 
> In fact in this:
> 
> * [Workarounds, limitations/missing features in ergocub 1.0 urdf #106](https://github.com/icub-tech-iit/ergocub-software/issues/106)
> 
> We reported that the missing gz-sim support is actually a known limitation, but I don't recall why

I think I dropped the integration of the gz-sim-yarp-plugins in the `1_0` models due to the reasons described in #234 , due to the fact that they would require some duplicated configuration files. I think that if we need to support also these models there is nothing strictly blocking.

Nicogene, 2025-07-17:
Here it is the SN000 in gz-sim, thanks to:
- #324 

[image]

PR #324 (2025-07-17): Refactor SN000 yaml
It remove duplications, it fixes #322, it fixes #234, it fixes #177, it fixes #281