mesh-iit/ergocub-software #294: Inconsistent l_elbow and r_elbow limits between ergoCub URDF and ergoCub documentation

traversaro, 2025-03-20:
Hello everyone! 

@fiorisi and @maggia80 noticed an interesting inconsistencies between the limits of ergoCub URDF and the actual (and documented) ergoCub limits.

In particular, in the documentation (https://icub-tech-iit.github.io/documentation/ergoCub1/ergoCub1-joints/#joint-3-elbow) the HW limits are [-5,+120], while in the URDF they are [-3,+75], see https://github.com/icub-tech-iit/ergocub-software/blob/724b02e2c5741e5861d68098149228ccc936a844/urdf/creo2urdf/data/common/ERGOCUB_joint_all_parameters.csv#L27 .

Interestingly, the limits on the robots-configuration side are consistent with the documentation (see https://github.com/robotology/robots-configuration/blob/3be388a8bbb7e1b588657ec7ac24feb8509a1c33/ergoCubSN002/hardware/mechanicals/left_arm-eb4-j2_3-mec.xml#L23-L24), so the bug seems to be isolated on the URDF side.

traversaro, 2025-03-20:
**TL;DR:**

**The limits of the elbow joints on URDF are wrong and much smaller then the documented one.**

**The problem probably affects the simulated ergoCub models and any application that only uses the URDF model of the ergoCub.**

**Real-world ergoCub application that retrieves the joint limits from the robot via the `yarp::dev::IControlLimits` YARP interfaces should not be affected by this. However, real robot applications that gets their limits from the URDF file, either directly or indirectly, for example via the `iDynTree::Joint::getPosLimits`, `iDynTree::Joint::getMinPosLimit` or  `iDynTree::Joint::getMinPosLimit` methods may be affected.**

traversaro, 2025-03-20:
Apparently the problem is not limited to the elbow joint, more updates will follow.

fiorisi, 2025-03-20:
| Joint                                  | [Documentation](https://github.com/icub-tech-iit/documentation/blob/master/docs/ergoCub1/ergoCub1-joints.md) HW ROM (°)    | [CSV](https://github.com/icub-tech-iit/ergocub-software/blob/master/urdf/creo2urdf/data/common/ERGOCUB_joint_all_parameters.csv) HW ROM (°)                              | Difference Details                                   |
|----------------------------------------|------------------------|---------------------------------------------|------------------------------------------------------|
| **Neck Pitch**                         | –45 / +22              | –30 / +22                                   | Lower limit is 15° less negative in CSV              |
| **Shoulder Yaw (r/l)**                  | –52 / +82              | –50 / +80                                   | Both limits are 2° less extreme in CSV               |
| **Elbow (r/l)**                        | –5 / +120              | –3 / +75                                    | Lower limit: 2° less negative; Upper limit: 45° lower  |
| **Index Adduction/Abduction (r/l)**     | +6 / +20               | 0 / 15                                      | Lower limit 6° lower; upper limit 5° lower           |
| **Torso Roll**                         | –25 / +25              | –23 / +23                                   | Both limits are 2° less extreme in CSV               |
| **Torso Pitch**                        | –20 / +65              | –18 / +45                                   | Lower: 2° less negative; Upper: 20° lower in CSV      |
| **Torso Yaw**                          | –45 / +45              | –43 / +43                                   | Both limits are 2° less extreme in CSV               |
| **Hip Pitch (r/l)**                    | –44 / +110             | –43 / +103                                  | Lower: 1° difference; Upper: 7° lower in CSV         |
| **Hip Roll (r/l)**                     | –35 / +111             | –16 / +109                                  | Lower limit differs by 19°; upper by 2°              |
| **Hip Yaw (r/l)**                      | –81 / +81              | –79 / +79                                   | Both limits are 2° less extreme in CSV               |
| **Knee (r/l)**                         | –105 / +7              | –104 / +6                                   | Both limits differ by 1°                             |

NOTE:
I think that we need to check the limits on the documentation page for the wrist and the Index Adduction/Abduction.

traversaro, 2025-03-20:
> And I didn't find changes since a while, I remember that I started from the limits of the icub3 urdf, is it possible that the limits has been changed in robots-configuration and we didn't port it to the model?

Let's track a bit the history of other repos:

`robots-configuration` : 
* The elbow max limit started at 105 in https://github.com/robotology/robots-configuration/pull/343 at March 2022, while the URDF file was added in December 2022 (see https://github.com/icub-tech-iit/ergocub-software/pull/3)
* In robots-configuration, the limit of the elbow was raised from 105 to 120 in https://github.com/robotology/robots-configuration/pull/440 in February 2023

`documentation` : 

The joint limit in documentation was always 120, but it was added later w.r.t. to the robots-configuration and ergocub-software, see https://github.com/icub-tech-iit/documentation/pull/297 added in August 2023.

traversaro, 2025-03-20:
For the software limits in the docs, we may need to also double check the software limits of the gazebo simulation, that are specified in `[LIMITS]` groups in init files: https://github.com/search?q=repo%3Aicub-tech-iit%2Fergocub-software+%5BLIMITS%5D+language%3AINI&type=code . On the real robot discussing with @fiorisi we tought that the software limits are the one specified in https://github.com/robotology/robots-configuration/blob/3be388a8bbb7e1b588657ec7ac24feb8509a1c33/ergoCubSN002/hardware/motorControl/left_arm-eb4-j2_3-mc.xml#L15-L16, but looking more in details it seems that a 2 degrees is also always removed to the limits in https://github.com/robotology/icub-main/blob/1c91cf2c368002e95c1f471d77d46c24f958c343/src/libraries/icubmod/embObjMotionControl/embObjMotionControl.cpp#L3367-L3368 , so we may also need to double check those.

traversaro, 2025-03-20:
Thanks @fiorisi ! @Nicogene I guess we need to coordinate for also updating the limits in the csv?

fiorisi, 2025-03-20:
I have submitted another [PR in the documentation](https://github.com/icub-tech-iit/documentation/pull/401) and I have updated accordingly the [table](https://github.com/icub-tech-iit/ergocub-software/issues/294#issuecomment-2740693380).

[image]

Nicogene, 2025-03-21:
The SN001/2 limits should be fixed by:
- https://github.com/icub-tech-iit/ergocub-software/pull/296

I need to understand how port this patch also on SN000 that we are updating it by hand

traversaro, 2025-03-21:
> now all ergocub should have the correct limits

What is the situation of software limits in the Gazebo .ini files? If those are still inconsistent we can open an issue for that?

Nicogene, 2025-03-21:
Similarly, I opened a PR in `r1-models` for porting also there the @fiorisi changes:
- https://github.com/icub-tech-iit/r1-models/pull/7
- https://github.com/icub-tech-iit/r1-models/pull/8

Nicogene, 2025-03-21:
> > now all ergocub should have the correct limits
> 
> What is the situation of software limits in the Gazebo .ini files? If those are still inconsistent we can open an issue for that?

I was wondering the same, let's reopen it

Nicogene, 2025-03-21:
I uploaded a bunch of xml files of ergoCubSN000 to copilot and I asked to him to extract a table containing both sw and hw limits, and here is the result

| **Joint Name**       | **jntPosMin** | **jntPosMax** | **hardwareJntPosMin** | **hardwareJntPosMax** |
|----------------------|---------------|---------------|-----------------------|-----------------------|
| neck_pitch           | -40           | +22           | -45                   | +22                   |
| neck_roll            | -20           | +20           | -20                   | +20                   |
| neck_yaw             | -45           | +45           | -45                   | +45                   |
| camera_tilt          | -30           | +30           | -30                   | +30                   |
| l_shoulder_pitch     | -130          | +14           | -170                  | +35                   |
| l_shoulder_roll      | 0             | +130          | -15                   | +150                  |
| l_shoulder_yaw       | -50           | +80           | -52                   | +82                   |
| l_elbow              | -3            | +105          | -5                    | +120                  |
| l_thumb_oc           | 4             | 72            | 0                     | +90                   |
| l_index_oc           | 4             | 85            | 0                     | +90                   |
| l_middle_oc          | 4             | 85            | 0                     | +90                   |
| l_ring_pinky_oc      | 4             | 85            | 0                     | +90                   |
| l_thumb_add          | 0             | 85            | 0                     | +90                   |
| l_index_add          | 8             | 21            | 0                     | +30                   |
| l_wrist_yaw          | -90           | +90           | -92                   | +92                   |
| l_wrist_roll         | -60           | +50           | -62                   | +52                   |
| l_wrist_pitch        | -30           | +30           | -32                   | +32                   |
| l_hip_pitch          | -42           | +102          | -44.81                | +110.15               |
| l_hip_roll           | -15           | +108          | -35                   | +111.77               |
| l_hip_yaw            | -78           | +78           | -81.65                | +81.65                |
| l_knee               | -103          | +5            | -105.23               | +7.75                 |
| l_ankle_pitch        | -43           | +43           | -46.624               | +46.624               |
| l_ankle_roll         | -23           | +23           | -25.313               | +25.313               |
| r_shoulder_pitch     | -130          | +14           | -170                  | +35                   |
| r_shoulder_roll      | 0             | +130          | -15                   | +150                  |
| r_shoulder_yaw       | -50           | +80           | -52                   | +82                   |
| r_elbow              | -3            | +105          | -5                    | +120                  |
| r_thumb_oc           | 4             | 70            | 0                     | +90                   |
| r_index_oc           | 4             | 85            | 0                     | +90                   |
| r_middle_oc          | 4             | 85            | 0                     | +90                   |
| r_ring_pinky_oc      | 4             | 85            | 0                     | +90                   |
| r_thumb_add          | 0             | 85            | 0                     | +90                   |
| r_index_add          | 7             | 18            | 6                     | +20                   |
| r_wrist_yaw          | -90           | +90           | -92                   | +92                   |
| r_wrist_roll         | -60           | +50           | -62                   | +52                   |
| r_wrist_pitch        | -30           | +30           | -32                   | +32                   |
| r_hip_pitch          | -42           | +102          | -44.81                | +110.15               |
| r_hip_roll           | -15           | +108          | -35                   | +111.463              |
| r_hip_yaw            | -78           | +78           | -81.078               | +81.078               |
| r_knee               | -103          | +5            | -105.23               | +7.75                 |
| r_ankle_pitch        | -43           | +43           | -45.94                | +45.94                |
| r_ankle_roll         | -23           | +23           | -25.786               | +25.786               |
| torso_roll           | -23           | +23           | -25                   | +25                   |
| torso_pitch          | -18           | +45           | -20                   | +65                   |
| torso_yaw            | -43           | +43           | -45                   | +45                   |

We could add something in the documentation? In this way it would be easier to navigate them instead of searching them inside the XML files

cc @fiorisi @pattacini @traversaro

traversaro, 2025-03-21:
> We could add something in the documentation? In this way it would be easier to navigate them instead of searching them inside the XML files

If I am not wrong the docs already contain the SW limits?

Nicogene, 2025-03-21:
> > We could add something in the documentation? In this way it would be easier to navigate them instead of searching them inside the XML files
> 
> If I am not wrong the docs already contain the SW limits?

Yes we should understand if the limits that we store in the documentation are for a specific robot(SN000/1/2) or a specific version of that robot (V1_0/1/3), probably the latter makes more sense

Nicogene, 2025-03-21:
Interestingly I asked to copilot the differences between SN000 and SN001 and here is the result:

| **Joint Name**       | **SN000 jntPosMin** | **SN000 jntPosMax** | **SN001 jntPosMin** | **SN001 jntPosMax** |
|----------------------|---------------------|---------------------|---------------------|---------------------|
| l_ring_pinky_oc      | 4                   | 85                  | 0                   | 70                  |
| l_wrist_roll         | -60                 | +50                 | -20                 | +30                 |
| l_wrist_pitch        | -30                 | +30                 | -20                 | +20                 |
| r_ring_pinky_oc      | 4                   | 85                  | 0                   | 70                  |
| r_wrist_roll         | -60                 | +50                 | -20                 | +30                 |
| r_wrist_pitch        | -30                 | +30                 | -20                 | +20                 |
| r_hip_yaw            | -78                 | +78                 | -79.199             | +79.199             |

| **Joint Name**       | **SN000 hardwareJntPosMin** | **SN000 hardwareJntPosMax** | **SN001 hardwareJntPosMin** | **SN001 hardwareJntPosMax** |
|----------------------|-----------------------------|-----------------------------|-----------------------------|-----------------------------|
| l_ring_pinky_oc      | 0                           | +90                         | 0                           | +70                         |
| l_ankle_pitch        | -46.624                     | +46.624                     | -46.885                     | +46.885                     |
| r_ring_pinky_oc      | 0                           | +90                         | 0                           | +70                         |
| r_hip_yaw            | -81.078                     | +81.078                     | -79.199                     | +79.199                     |
| r_ankle_pitch        | -46.624                     | +46.624                     | -46.885                     | +46.885                     |

The most relevant changes seems relate to the wrist/hands, maybe because they mount different hands ? (mk5.0 mk5.1) cc @fiorisi 

I expect that SN001 and SN002 share the same limits

traversaro, 2025-03-21:
Beside the wrist/hands, the `r_hip_yaw` and `r_ankle_pitch` difference are fishy, especially if there is not the same different on the left part.

Nicogene, 2025-03-21:
> I expect that SN001 and SN002 share the same limits

I don't know if it is an attendible result but here are the differences between SN001 and SN002

| Joint Name       | Parameter           | SN001   | SN002   |
|------------------|---------------------|---------|---------|
| l_ring_pinky_oc  | jntPosMax           | +80     | +70     |
| r_ring_pinky_oc  | jntPosMax           | +80     | +70     |
| r_hip_yaw        | hardwareJntPosMin   | -79.808 | -79.199 |
| r_ankle_pitch    | hardwareJntPosMax   | +45.94  | +46.885 |

Nicogene, 2025-03-21:
Thanks to the help of copilot I can now change the ini accordingly, the only problem is how we address the differences between the wrist of SN000 and SN001/2, since the ini is shared between the robots, shall I need to create a separate ini for the two robots?

traversaro, 2025-03-21:
> Thanks to the help of copilot I can now change the ini accordingly,

Before changing the .ini I should check if the software limits are expected to be different or if there is an error in the robots configuration files. If for the wrist the version is indeed different, I would keep two separate ini files (and robotinterface files).

Nicogene, 2025-03-24:
The differences in the wrists respect the fact that on the SN000 there is mounted the mk2.0 and on the SN001/2 the mk2.1

- https://icub-tech-iit.github.io/documentation/wrists/wrist_mk2/

I will proceed then creating two different in files and fix all the other limits

fiorisi, 2025-03-24:
> I expect that SN001 and SN002 share the same limits

I've checked the snapshots and the SN001 and SN002 should have the same mechanics.

## SN001

[image]
[image]

## SN002

[image]
[image]

Nicogene, 2025-03-24:
Ok the software limits in the urdf had several errors (the most important is in the torso):
- https://github.com/icub-tech-iit/ergocub-software/pull/298

tagging the people that uses the these models

cc @GiulioRomualdi @mebbaid @PasMarra @S-Dafarra

PR #297 (2025-03-21): ergoCubV1: fix limits
See #295. #296

It fixes #294
commit 78b4151b (2025-03-24): Fix the SW limits of the ergocub models

Fixes #294
closed 2025-03-24 by Nicogene