mesh-iit/ergocub-software #299: Wrong chest meshpath in ergocub002.urdf

GiulioRomualdi, 2025-04-01:
When I load the ergocub002 urdf the chest mesh is not found. 
The path for the chest mesh might be wrong, see
- https://github.com/icub-tech-iit/ergocub-software/blob/ce6e32b8365a668fd53dba02c60ff013442ffbd3/urdf/ergoCub/robots/ergoCubSN002/model.urdf#L177
- https://github.com/icub-tech-iit/ergocub-software/blob/ce6e32b8365a668fd53dba02c60ff013442ffbd3/urdf/ergoCub/robots/ergoCubSN002/model.urdf#L186

As additional info, I changed the path from `sim_ecub_1-3_chest.obj` to `sim_ecub_1-1_chest.obj` and the `urdf` is correctly loaded!

Nicogene, 2025-04-01:
Mmm, very strange, I never experienced this kind of issue, for example, here:

- https://github.com/icub-tech-iit/ergocub-software/issues/259#issuecomment-2648598291

The mesh of the chest is correctly loaded, and if I am not _prendendo un abbaglio_ the file name is correct:
- https://github.com/icub-tech-iit/ergocub-software/blob/master/urdf/ergoCub/meshes/simmechanics/sim_ecub_1-3_chest.obj

traversaro, 2025-04-01:
> When I load the ergocub002 urdf the chest mesh is not found.

Which software are you using to load the URDF? Note that `sim_ecub_1-3_chest.obj` is created by Blender, while `sim_ecub_1-1_chest.obj` by MeshLab, I wonder if there are subtles differences of the produced `.obj`. Perhaps we could try to load the `sim_ecub_1-3_chest.obj` with MeshLab and save it again with MeshLab to "normalize" it?