mesh-iit/ergocub-software #300: ergoCub_X_minContacts makes crash gz-sim

Nicogene, 2025-04-17:
gz-sim seems to not like collision made by boxes with dimensions 0.

```
gz sim server: ./dart/dynamics/BoxShape.cpp:43: dart::dynamics::BoxShape::BoxShape(const Eigen::Vector3d&): Assertion `_size[0] > 0.0' failed.
Stack trace (most recent call last):
#31   Object "/lib/x86_64-linux-gnu/libruby-3.2.so.3.2", at 0x7f95df8a92c9, in rb_yield
#30   Object "/lib/x86_64-linux-gnu/libruby-3.2.so.3.2", at 0x7f95df8a4b39, in rb_vm_exec
#29   Object "/lib/x86_64-linux-gnu/libruby-3.2.so.3.2", at 0x7f95df8a15db, in
#28   Object "/lib/x86_64-linux-gnu/libruby-3.2.so.3.2", at 0x7f95df89d0fe, in
#27   Object "/lib/x86_64-linux-gnu/libruby-3.2.so.3.2", at 0x7f95df89a8ef, in
#26   Object "/usr/lib/x86_64-linux-gnu/ruby/3.2.0/fiddle.so", at 0x7f95da75cb13, in
#25   Object "/lib/x86_64-linux-gnu/libruby-3.2.so.3.2", at 0x7f95df8633bb, in rb_nogvl
#24   Object "/usr/lib/x86_64-linux-gnu/ruby/3.2.0/fiddle.so", at 0x7f95da75c43b, in
#23   Object "/lib/x86_64-linux-gnu/libffi.so.8", at 0x7f95da6e70bd, in ffi_call
#22   Object "/lib/x86_64-linux-gnu/libffi.so.8", at 0x7f95da6e43ee, in
#21   Object "/lib/x86_64-linux-gnu/libffi.so.8", at 0x7f95da6e7b15, in
#20   Object "/usr/lib/x86_64-linux-gnu/libgz-sim9-gz.so.9.0.0", at 0x7f95d9be742d, in runServer
#19   Object "/lib/x86_64-linux-gnu/libgz-sim9.so.9", at 0x7f95d98ef7b5, in
#18   Object "/lib/x86_64-linux-gnu/libgz-sim9.so.9", at 0x7f95d98fc2f6, in gz::sim::v9::SimulationRunner::Run(unsigned long)
#17   Object "/lib/x86_64-linux-gnu/libgz-sim9.so.9", at 0x7f95d98fba21, in gz::sim::v9::SimulationRunner::Step(gz::sim::v9::UpdateInfo const&)
#16   Object "/lib/x86_64-linux-gnu/libgz-sim9.so.9", at 0x7f95d98f836c, in gz::sim::v9::SimulationRunner::UpdateSystems()
#15   Object "/usr/lib/x86_64-linux-gnu/gz-sim-9/plugins/libgz-sim-physics-system.so", at 0x7f9598554a59, in
#14   Object "/usr/lib/x86_64-linux-gnu/gz-sim-9/plugins/libgz-sim-physics-system.so", at 0x7f9598558c76, in
#13   Object "/usr/lib/x86_64-linux-gnu/gz-sim-9/plugins/libgz-sim-physics-system.so", at 0x7f95985d3f44, in
#12   Object "/usr/lib/x86_64-linux-gnu/gz-physics-8/engine-plugins/libgz-physics-dartsim-plugin.so", at 0x7f94b41bfa16, in virtual thunk to gz::physics::dartsim::SDFFeatures::ConstructSdfNestedModel(gz::physics::Identity const&, sdf::v15::Model const&)
#11   Object "/usr/lib/x86_64-linux-gnu/gz-physics-8/engine-plugins/libgz-physics-dartsim-plugin.so", at 0x7f94b41bea52, in gz::physics::dartsim::SDFFeatures::ConstructSdfModelImpl(unsigned long, sdf::v15::Model const&)
#10   Object "/usr/lib/x86_64-linux-gnu/gz-physics-8/engine-plugins/libgz-physics-dartsim-plugin.so", at 0x7f94b41beab1, in gz::physics::dartsim::SDFFeatures::ConstructSdfModelImpl(unsigned long, sdf::v15::Model const&)
#9    Object "/usr/lib/x86_64-linux-gnu/gz-physics-8/engine-plugins/libgz-physics-dartsim-plugin.so", at 0x7f94b41b62a9, in gz::physics::dartsim::SDFFeatures::FindOrConstructLink(std::shared_ptr<dart::dynamics::Skeleton> const&, gz::physics::Identity const&, sdf::v15::Model const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
#8    Object "/usr/lib/x86_64-linux-gnu/gz-physics-8/engine-plugins/libgz-physics-dartsim-plugin.so", at 0x7f94b41b9834, in gz::physics::dartsim::SDFFeatures::ConstructSdfLink(gz::physics::Identity const&, sdf::v15::Link const&)
#7    Object "/usr/lib/x86_64-linux-gnu/gz-physics-8/engine-plugins/libgz-physics-dartsim-plugin.so", at 0x7f94b41b9eb6, in gz::physics::dartsim::SDFFeatures::ConstructSdfCollision(gz::physics::Identity const&, sdf::v15::Collision const&)
#6    Object "/usr/lib/x86_64-linux-gnu/gz-physics-8/engine-plugins/libgz-physics-dartsim-plugin.so", at 0x7f94b41b7eb7, in
#5    Object "/lib/x86_64-linux-gnu/libdart.so.6.13", at 0x7f9494484cdc, in dart::dynamics::BoxShape::BoxShape(Eigen::Matrix<double, 3, 1, 0, 3, 1> const&)
#4    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f95df2b1516, in __assert_fail
#3    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f95df29e81a, in
#2    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f95df29e8fe, in abort
#1    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f95df2bb27d, in gsignal
#0    Object "/lib/x86_64-linux-gnu/libc.so.6", at 0x7f95df314b2c, in pthread_kill
```

traversaro, 2025-04-18:
I may be missing something, but why we need boxes with dimensions 0? The SDF and URDF spec do not explicitly constraint the box dimension to be >0 (see https://github.com/gazebosim/sdformat/blob/493a8c9fa4fb1cb5818225ecf3f77ea90aa76423/sdf/1.12/box_shape.sdf and https://github.com/gazebosim/sdformat/blob/493a8c9fa4fb1cb5818225ecf3f77ea90aa76423/sdf/1.12/box_shape.sdf), but that seems a reasonable assumption. If we have boxes of dimensions 0, can't we just remove them?

Nicogene, 2025-04-30:
> but that seems a reasonable assumption. If we have boxes of dimensions 0, can't we just remove them?

Good point, we have boxes with size zero because right now it is the only way to disable collision via `assignCollisionGeometry` of creo2urdf, maybe we can just add a parameter in the yaml for specify the list of links of which we want to disable the collision, and use `assignCollisionGeometry` for assigning simplified geometries (cc @mfussi66)

traversaro, 2025-04-30:
> Good point, we have boxes with size zero because right now it is the only way to disable collision via `assignCollisionGeometry` of creo2urdf, maybe we can just add a parameter in the yaml for [...]

Yes, I think it make more sense to handle this at the creo2urdf level rather then at gz-sim level.

Nicogene, 2025-05-28:
xref:
- https://github.com/icub-tech-iit/r1-models/pull/25

Nicogene, 2025-05-28:
| Simulation Environment | ergoCubGazeboV1_1_minContacts | ergoCubGazeboV1_3_minContacts | R1Mk3Gazebo_minContacts |
|------------------------|-------------------------------|-------------------------------|--------------------------|
| gazebo classic         |  [image]                      |  [image]                        | 

[image]

                       |
| gz-sim                 | 

[image]

                            | 

[image]

                   | 

[image]

                    |

PR #314 (2025-05-28): Change minContacts addon in order to use empty instead of a box of 0 dimension
It fixes #300